ArduPilot/ardupilot #11380: [Feature request] Support Holybro Kakute F7 mini SPI Flash (Blackbox)

bys1123, 2019-05-20:
### Feature request

Holybro just release their new FC, Kakute F7 mini, it's using a 16MB (128Mbit) SPI flash instead of TF card.
https://shop.holybro.com/kakute-f7-mini_p1133.html

**Platform**
[√] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
flash model: W25Q128FVSIG
SPI pin:
[image]

maroquio, 2019-08-03:
Would be awesome to have this feature enabled. Kakute F7 Mini is 4g weight and have a lot of features. I always got “Bad logging” when flying my under 250g drone, and I think it is something related to the sd-card. With an internal memory, I think this error would be gone.

R-Lefebvre, 2019-11-29:
Closing issue as logging appears to be working in Master now.

closed 2019-11-29 by R-Lefebvre