ArduPilot/ardupilot #11385: Don't allow clearing mission by uploading zero items

peterbarker, 2019-05-21:
### Feature request

Consider not allowing clearing mission by stealth by uploading zero items.

Currently if you use the mavlink option to clear the mission then you will be rejected if the Mission is currently running.   But if you upload an empty mission we will simply wipe it.

A PR will be created for discussion where we special-case uploading zero items and treat it as a clear - if you are running the mission it  will be rejected.

Note that uploading zero items is no different to uploading 1 item if you are currently running misson item 5!  This is a reason such a PR may be rejected.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

OXINARF, 2019-05-21:
I understand your reasoning but, as you said, I would argue that uploading zero items is the same as uploading any other number.

I have a branch for adding an option to not allow uploads while the mission is running (and not allow entering Auto while uploading is running, which also prevents half uploaded missions from running). It's based on Copter 3.5 and only has support for Copter - I've recently tried to rebase it but was out of time to finish it.

peterbarker, 2024-07-31:
Too old - and a reasonable argument was made.

closed 2024-07-31 by peterbarker