ArduPilot/ardupilot #11398: AP_Mount: Solo gimbal overshooting in ChibiOS

Pedals2Paddles, 2019-05-23:
### Bug report
**Issue details**
In master chibios, the Solo gimbal's tilt control is overshooting badly.  

On the old FW, you can see the gimbal's EKF and target are in lock-step with eachother.  And you can see the rate controller smoothly comes in and out as it should in time with the the gimbal tilt reaching its target.  _It also continues to work correctly on 3.6 stable Nuttx_.
[image]

On master ChibiOS, you can see the target and EKF are no longer on the same page.  And you can see the rate controller is overshooting along with the EKF.  The gimbal is performing as the EKF and rate control indicate, overshooting its target and then coming back to it.
[image]

Things I have already tried while also discussing with JC:
- DMA_NOSHARE on the serial 4 UART
- Undoing the private mavlink channel patch
- Undoing the HAVE_PAYLOAD_SPACE patch

I don't see any commits between 3.6 and now in the mount library that would do this. The only functional difference is Nuttx vs ChibiOS that I can see. Unless it is somewhere else in AP.  Logs attached below.

As a hack, I found that derating the rate output by 75% slows it down enough to work properly. But this is obviously not a solution. However it demonstrates that somewhere along the line, the EKF is 75% ahead of where it should be.  Or something like that.  IDK.

**Version**
Copter 3.7-dev

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
Cube

**Logs**
[Solo Gimbal Logs.zip](https://github.com/ArduPilot/ardupilot/files/3209921/Solo.Gimbal.Logs.zip)

jschall, 2019-08-21:
Need to make sure that the SoloGimbalEKF is using all of the delta angle data coming out of the gimbal.

tridge, 2019-08-27:
Matt, for a bench test (no flying) can you try changing this line:
https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/Scheduler.h#L29
to change APM_UART_PRIORITY to 181 instead of increasing the buffer size.
The reason this is puzzling is we already have a 1k buffer, and the change that fixed the issue moved it to 1.5k. The 1k comes from 512 bytes in the chibios serial driver for non-DMA UARTs, plus 512 in HAL_ChibiOS

tridge, 2019-08-27:
it looks like this issue was caused by interrupt latency due to the i2c interrupt storm that is now fixed. Closing this, but please re-open if it turns out it isn't fully fixed