ArduPilot/ardupilot #11406: AP_Soaring: Wiggles when entering loiter

samuelctabor, 2019-05-24:
### Bug report

**Issue details**

Plane exhibits wiggle when entering LOITER after thermal detected. Particularly evident in Marco Robustini's log attached.

**Version**
3.9.8 (6ea22)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
 ASW28 4 mt. wingspan

**Hardware type**
Pixhawk 1

**Logs**
[2019-05-15.16-14-05.zip](https://github.com/ArduPilot/ardupilot/files/3217541/2019-05-15.16-14-05.zip)

Target roll angle very noisy
[image]

Airspeed control isn't bad
[image]

Quite pitchy
[image]

Estimated thermal position movement appears reasonable after initial transient
[image]

Updated code with a number of fixes available here https://github.com/samuelctabor/ardupilot/tree/SoaringImprovements-398

Compiled for Pixhawk 1 here
[arduplane.zip](https://github.com/ArduPilot/ardupilot/files/3217617/arduplane.zip)

robustini, 2019-05-24:
Thanks Sam!
Soon as possible I try and i leave you feedback and df log.

robustini, 2019-05-26:
@samuelctabor please, could you compile the firmware in APJ format?

samuelctabor, 2019-05-26:
Hi Marco,

Here it is [arduplane-apj.zip](https://github.com/ArduPilot/ardupilot/files/3220303/arduplane-apj.zip)

Should be commit 291ad.

robustini, 2019-05-28:
Hi @samuelctabor , i've tried the new code with my Elipsoid glider and there is a big problem, both in FBWB and AUTO many times it activates the motor, while with the official 3.9.8 it does what it should do.
Here two logs, one with your fix and the other with the official 3.9.8.
I'm sorry for the low bitmask.

[logs.zip](https://github.com/ArduPilot/ardupilot/files/3228484/logs.zip)

samuelctabor, 2019-05-29:
Hi @robustini ,

It's related to the ARSPD_USE=2 setting, I haven't tested this before. When throttle is >0 the airspeed sensor is not used, and this calls a different function in the TECS, which does not set throttle to zero like the usual with-airspeed function.

I'll fix and recompile today.

Sam

robustini, 2019-05-29:
Thanks Sam!
Meanwhile I'm trying Frigateboard on the sim, it's not bad at all, besides the SOAR part provides a myriad of parameters with respect to the Ardupilot code.

https://youtu.be/ZpCTVLUqy48

samuelctabor, 2019-05-29:
Hi Marco,

Updated firmware - should be commit c9322. Fixes throttle issue with ARSPD_USE=2 and aligns methods with PR https://github.com/ArduPilot/ardupilot/pull/9643
 [arduplane-c9322.zip](https://github.com/ArduPilot/ardupilot/files/3233328/arduplane-c9322.zip)

Yes there are a lot of parameters in the FrigateBird POMDP method. I started some work to simplify a little - https://github.com/samuelctabor/ardupilot/tree/Soar-POMDP-master - a work in progress and not flight tested.

robustini, 2019-05-30:
Thanks Sam, if I can I try it in the air this afternoon.
I would have an implementation to ask you, which I think there is in the POMDP, define a radius where if the glider reaches it during Loiter it does not return to the home but does something else.
For example if in AUTO it goes into the Loiter to follow a thermal and reaches the max radius could do that it passes to the next waypoint, or it takes the home as wp and when it gets there it goes to the next waypoint.
In FBWB you could do the same thing, or create a sort of "default angle" where when it reaches the radius it goes back inwards with a definable radius (for example 45 °).
This is to ensure that the glider always continues to autonomously look for the thermals without ever performing an RTL.

robustini, 2019-05-31:
Hi @samuelctabor, I'm trying your changes and it seems to me that it goes much much much better!
However there is a problem, in SOAR it never activates the motors in any case, for example when it adds the SOAR_ALT_MIN and goes to RTL, or when you go to AUTO.
However today day with scary thermals, look at the log!
I await the fix of the throttle problem.
[logs_new.zip](https://github.com/ArduPilot/ardupilot/files/3241650/logs_new.zip)

Naterater, 2019-06-02:
This happens to all of my planes, and I don't use soaring. Do you mind if I remove the AP_Soaring prefix?

samuelctabor, 2019-06-02:
Hi Marco,

New version with throttle issue fixed 
[arduplane-31ade.zip](https://github.com/ArduPilot/ardupilot/files/3245271/arduplane-31ade.zip)

samuelctabor, 2019-06-02:
Hi Nate,

I'm still concerned that there may be a regression specific to soaring - the target roll angle in the OP looks off to me - compare to Marco's recent flights using my dev branch 

[image]

The target still moves around and even reverses, as the target location is updated, but not very high frequency as in the stable release.

robustini, 2019-06-03:
> 
> 
> Hi Marco,
> 
> New version with throttle issue fixed
> [arduplane-31ade.zip](https://github.com/ArduPilot/ardupilot/files/3245271/arduplane-31ade.zip)

Hi Sam, thanks for the fix, I tried it now with X-Plane 11 and SITL using your branch and it seems to work regularly, as soon as I can I try it at the airfield.

samuelctabor, 2019-06-03:
Sounds good Marco. You could try setting your WP_LOITER_RAD smaller as well. 20-30m should work for your aircraft at 12m/s.

robustini, 2019-06-04:
Hi @samuelctabor, I noticed this problem in your 3.9.8 SOAR version, the FBWB speed is not maintained regularly in the simulator, while with the Ardupilot Master it works perfectly.
Same parameters... is this normal?
I set the SOAR_VSPEED to 50 so that it maintains the FBWB without going into the Loiter, in the first part of the video I do the test with Ardupilot Master, in the second (at 02:00) with your "SoaringImprovements-398" branch.
Watch when I perform FBW MIN changes in both tests.
What do you think?

https://youtu.be/WEfFOkThVHE

robustini, 2019-06-05:
@samuelctabor i've another request regarding the logic of the SOAR: if the SOAR is active in FBWB and goes to Loiter because it finds a thermal it does not seem correct to me that when it considers it vanished you switch RTL, it should do so only if you have reached the SOAR_AL_MIN, should instead return to FBWB.
Now:

"If the previous mode was FBWB or CRUISE, and thermalling ended due to reaching SOAR_ALT_MIN, RTL will be triggered instead"

It's fine, but not if the glider is inside the SOAR ALT range.

robustini, 2019-06-05:
In "Soaring.cpp", line 66 and 67, please consider this changes:

                gcs().send_text(MAV_SEVERITY_INFO, "Soaring: Thermal ended, restoring FBWB");
                set_mode(FBWB, MODE_REASON_SOARING_THERMAL_ESTIMATE_DETERIORATED);

robustini, 2019-06-05:
In "soaring.cpp", line 66 and 67, please consider this changes, already tested in the sim and work fine:

                gcs().send_text(MAV_SEVERITY_INFO, "Soaring: Thermal ended, restoring FBWB");
                set_mode(mode_fbwb, MODE_REASON_SOARING_THERMAL_ESTIMATE_DETERIORATED);

robustini, 2019-06-05:
In "soaring.cpp", line 66 and 67, please consider this changes, already tested in the sim and work fine:

gcs().send_text(MAV_SEVERITY_INFO, "Soaring: Thermal ended, restoring FBWB");
g2.soaring_controller.init_thermalling();
set_mode(mode_fbwb, MODE_REASON_SOARING_THERMAL_ESTIMATE_DETERIORATED);

robustini, 2019-06-05:
I found the problem, just reinitialize the cruising after returning to FBWB, now the mode rotation work fine:

                gcs().send_text(MAV_SEVERITY_INFO, "Soaring: Thermal ended, restoring FBWB");
	        plane.g2.soaring_controller.init_cruising();
                set_mode(mode_fbwb, MODE_REASON_SOARING_THERMAL_ESTIMATE_DETERIORATED);

magicrub, 2019-06-05:
hey @robustini take a look at this PR

magicrub, 2019-06-05:
>plane.g2.soaring_controller.init_cruising();
set_mode(mode_fbwb, MODE_REASON_SOARING_THERMAL_ESTIMATE_DETERIORATED);

Hmm, that should have no effect because that happens already in the set_mode for modes [fbwb](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/mode_fbwb.cpp#L12), [cruise](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/mode_cruise.cpp#L14), and [auto](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/mode_auto.cpp#L27).

robustini, 2019-06-05:
Absolutely agree with you, yet I have tried and so it works, maybe it's just a coincidence.