ArduPilot/ardupilot #11409: The FlightController "F4BY" don‘t have SERVORAIL_VCC_SENS

cesforchina, 2019-05-26:
I have a F4BY,I found that it does not show the servo mains voltage.But I saw in the code that there is a definition of the servo voltage port
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/F4BY/hwdef.dat
"PC5 FMU_SERVORAIL_VCC_SENS ADC1 SCALE(2)"
The schematic also shows that there is this adc port，I hope you can fix this bug, thanks.

sh83, 2019-06-19:
Some time ago some ebay/aliexpress/bg sellers sell so called "Ardupilot PixConcise Compatible Pixhawk Flight Controller" . It have no relation to the pixhawk and it is actually f4by clone with some modifications. It uses different imu and compass.
So you are probably use clone what do not have vservo connected. 
So nothing can be done in software.

cesforchina, 2019-06-19:
Thank you for your reply. In fact, my F4BY is designed according to the F4BY2.1.5 schematic, so I am sure the problem comes from the software side. I also tried to modify the description of V_servo to be the same as fmuV3 in ChiBiOS.hal, but Servo_voltage Still shown as 0000 (even without ADC error)

sh83, 2019-06-19:
Yes, looks like currently servo voltage sensed only on boards with iomcu.

OXINARF, 2019-06-20:
@kozinalexey

kozinalexey, 2019-06-20:
https://github.com/ArduPilot/ardupilot/blob/6a27e8585a2d20fab2bee68a0d91ef278653d366/libraries/AP_HAL/board/chibios.h#L24

kozinalexey, 2019-06-23:
https://github.com/kozinalexey/ardupilot/commit/b66720796c4199e05ca7675f8c2f52f3d3c3c6c5

kozinalexey, 2019-07-02:
Hi, fix merged to master

closed 2019-07-02 by sh83