ArduPilot/ardupilot #11413: AUTO with no mission items causes crash

AndKe, 2019-05-26:
### Bug report

if AUTO mode is engaged with no mission items, but MIS_TOTAL indicating some,.. it will result in crash.
 

To reproduce in SITL (ArduCopter):
wp list   // observe no items
param set MIS_TOTAL 40    //set some items, as if restoring params from backup
rc 3 1500
poshold
arm throttle
takeoff 20
AUTO  // now observe that AUTO is never confirmed, link goes down..

**Version**
3.6.8 & Current master

**Platform**
[x ] Copter
- not sure about the others.

**Airframe type**
N/A

**Hardware type**
SITL/Pixhawk

**Logs**
Please see screenshot... 

[image]

WickedShell, 2019-05-26:
This is unfortunately fundamentally related to the fact that you should never manipulate `MIS_TOTAL` parameter. (Indeed it shouldn't be a parameter, but is). We shouldn't crash in that case though. I've seen this confuse GCS's on downloading missions before as well. I haven't investigated the root cause on the vehicle side, but we should be including protection against this, since we've exposed this as a way to shoot yourself in the foot.

rmackay9, 2019-05-27:
yes, indeed, we should protect against this, thanks for the report!

peterbarker, 2019-05-28:
Concept 1: ignore param-sets of `MISSION_CMD` (and a select list of other parameters the GCS should not be touching; just pretend they were set)
Concept 2: mark these parameters as being for a specific non-sensical frame - thus being excluded from the downloaded parameter lists
Concept 3: add a new `AP_PARAM_HIDDEN` attribute and use it to hide parameters like the FRAME stuff does

Pedals2Paddles, 2019-05-28:
Maybe this is a silly question. But why does this exist at all then? If it's is going to be hidden or locked out, doesn't even have a purpose?

AndKe, 2019-05-28:
Solution suggestions: 
1:  Just ignore any mission item with Command ID=0  , jump ring to next mission item,  parsing up to MISSION_TOTAL

2: Given that we agree that CID of 0 is invalid, then just refuse to enter AUTO mode, or stop if that is found.

CID(and other values) set to zero is the normal thing to expect in an empty flash.

WickedShell, 2019-05-28:
> Maybe this is a silly question. But why does this exist at all then? If it's is going to be hidden or locked out, doesn't even have a purpose?

Minimum effort. Parameters are the default/lazy way to persistently store data, so they will be used even if they introduce vulnerabilities like this.

@AndKe Unfortunately you can also have problems by clearing the mission then setting the total back up, at that point you have valid, but unexpected mission items uploaded, which isn't good either. The control that lets you get into the problem is the worst part.

AndKe, 2019-06-01:
Maybe the solution should not be to handle crazy configuration mismatch, but rather not to fall out of the sky when executing a mission item that consists of zero in every field?
if such item were ignored, nothing bad would happen.

rmackay9, 2019-06-04:
This was discussed on the dev call and I think we decided that we would go with @peterbarker's 1st idea:

> "Concept 1: ignore param-sets of MISSION_TOTAL (and a select list of other parameters the GCS should not be touching; just pretend they were set)"

magicrub, 2019-06-04:
So basically, implement the @volatile Param field metadata and then inhibit writes to them. Also, works be nice to have it as a bitfield where we can have read-only, calibration, ect.. that would help the GCS when saving to a file where it could skip calibrations, for example.

rmackay9, 2019-06-04:
@magicrub, I think the solution would be much more specific.. so it would be a special case just for this one parameter.

peterbarker, 2019-06-04:
@rmackay9 I did already create the PR for this - and it is specific to this one problematic parameter.  

GCSs should probably already run a blacklist - as @WickedShell has pointed out (and I realised while writing it) there's no indication the set has failed until you read it back.

Pedals2Paddles, 2019-06-04:
Would implementing @AndKe's suggestion be wise as well?  If a mission item of all zeros makes bad things happen, that should probably be checked and inhibited regardless of the parameter solution.

commit 4f9d1ada (2019-08-21): GCS_MAVLink: deny setting MIS_TOTAL parameter

Closes #11413

GCS_MAVLink: conform to parameter protocol requirements while ignoring sets

GCS_MAVLink: use internal-use-only param bit
closed 2019-08-21 by peterbarker