ArduPilot/ardupilot #11418: AC_WPNav:Formula: Is "_slow_down_dist = speed_cms * speed_cms /(4.0f * accel_cmss)" correct?

BreederBai, 2019-05-27:
According to the kinematic formula, v1^2-v0^2=2ax, where v0 is the initial velocity. When v0 is 0, x = v1^2/2a, but the calculation of the deceleration distance in the program seems to be x = v^ 2/4a, this seems to be contrary to the kinematics formula. Is this a mistake?
```
/// calc_slow_down_distance - calculates distance before waypoint that target point should begin to slow-down assuming it is travelling at full speed
void AC_WPNav::calc_slow_down_distance(float speed_cms, float accel_cmss)
{
	// protect against divide by zero
	if (accel_cmss <= 0.0f) {
		_slow_down_dist = 0.0f;
		return;
	}
    // To-Do: should we use a combination of horizontal and vertical speeds?
    // To-Do: update this automatically when speed or acceleration is changed
    _slow_down_dist = speed_cms * speed_cms / (4.0f*accel_cmss);
}

/// get_slow_down_speed - returns target speed of target point based on distance from the destination (in cm)
float AC_WPNav::get_slow_down_speed(float dist_from_dest_cm, float accel_cmss)
{
    // return immediately if distance is zero (or less)
    if (dist_from_dest_cm <= 0) {
        return WPNAV_WP_TRACK_SPEED_MIN;
    }

    // calculate desired speed near destination
    float target_speed = safe_sqrt(dist_from_dest_cm * 4.0f * accel_cmss);

    // ensure desired speed never becomes too low
    if (target_speed < WPNAV_WP_TRACK_SPEED_MIN) {
        return WPNAV_WP_TRACK_SPEED_MIN;
    } else {
        return target_speed;
    }
}
```

lthall, 2019-08-03:
No, this is not a mistake. We accelerate and decelerate the leash end point at twice the acceleration because the leash reduces the acceleration of the aircraft.

closed 2019-08-03 by lthall

magicrub, 2019-08-04:
Perhaps we could add a comment in the source to explain that for next time?

lthall, 2019-08-04:
Yeh, the problem with this navigation code is if I did that with all the little twists and turns like this there would be as many lines of comments as code.

The beauty of S-Curve implementation is that it will be 100% kinematically correct and not need all these little subtle tricks to make it work. The S-Curve math will be extremely complex but that will be in a separate library. The navigation part will be extremely simple in comparison and infinitely more readable than our current implementation!