ArduPilot/ardupilot #11421: Provide support for D-term FFT logging

andyp1per, 2019-05-27:
### Feature request

Provide support for analyzing the frequency characteristics and noise of the D-term, both pre- and post-filter. Betaflight allows this and it's quite key in tuning the D-term filters for small quads.
I have a fix for this.

**Describe the solution you'd like**
Log options similar to what are available for gyro frequency logging
**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2019-05-27:
This is fixed by #11422

andyp1per, 2019-06-16:
Proposed Design
- Refactor common parts of BatchSampler into AP_Logger/AP_BatchSampler
- Deprecate INS_LOG_BAT_LGIN, INS_LOG_BAT_LGCT, INS_LOG_BAT_CNT and use LOG_BAT_LGIN, LOG_BAT_LGCT, LOG_BAT_CNT instead
- Change AP_BatchSampler::sample() to:
  - uint8_t _instance
  - AP_BatchSampler::CONTROL_POINT _type
  - uint64_t sample_us
  - const Vector3f &_sample
  where instance is the sensor instance for gyro/accels and P==0, I==1, D==2 for ATC
  where CONTROL_POINT can be
  - IMU_SENSOR_OUT_ACCEL
  - IMU_SENSOR_OUT_GYRO
  - IMU_FILTER_OUT_ACCEL
  - IMU_FILTER_OUT_GYRO
  - ATC_SIGNAL_IN
  - ATC_FILTER_OUT
  where sample is [x, y, z] or [roll, pitch, yaw]
- Create subclasses of AP_BatchSampler in AC_AttitudeControl and AP_InertialSensor for the instance specific code. 
- Have a single push_data_to_log periodic task

This means all the messages will be of the same type. An alternative would be to keep ISDB and ISDH for INS and introduce ATDB and ATDH for AttitudeControl, but sharing as much code as possible.

IamPete1, 2022-09-11:
https://github.com/ArduPilot/ardupilot/pull/11422#issuecomment-900456381

closed 2022-09-11 by IamPete1