ArduPilot/ardupilot #11428: Pixhawk1 or fmuv2 can not run.

LeiOuYang, 2019-05-28:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**
Eclipse and waf bulid the code(base the python3.6m).Pargram can not run: The led of F427 always flashing.

**Version**
Copter3.7.0-dev

**Hardware type**
Pixhawk1

rmackay9, 2019-05-28:
Looks like a developer question so feel free to ask in the [developer gitter chat room](https://gitter.im/ArduPilot/ardupilot).

closed 2019-05-28 by rmackay9

LeiOuYang, 2019-05-29:
OK！Thank you!

LeiOuYang, 2019-05-29:
> 
> 
> Looks like a developer question so feel free to ask in the [developer gitter chat room](https://gitter.im/ArduPilot/ardupilot).

OK! Thank you!