ArduPilot/ardupilot #11431: Matek F405-CTR FrSky telemetry does not work

twistedwings, 2019-05-28:
ArduCopter 3.7.0

inverted S.Port (A/B) from XSR connected to Tx3/Rx3
[image]

SERIAL1_BAUD 57
SERIAL1_OPTIONS 4
SERIAL1_PROTOCOL 4

No sensors discovered

Also tried Tx5/Rx5

IamPete1, 2019-05-30:
You also need a diode to convert from half duplex to full, and you don't have to discover sensors. This is the best place to get help setting it up, https://discuss.ardupilot.org/t/some-soldering-required/27613

closed 2019-05-30 by IamPete1

twistedwings, 2019-05-30:
Why do I need a diode if halfduplex (SERIAL1_PROTOCOL 4) is configured?
The same configuration works perfect on a F405-Wing.