ArduPilot/ardupilot #11435: Copter: 3.7-dev allows forced arming in Loiter without a good position estimate

rmackay9, 2019-05-29:
Similar to [this issue found for Rover](https://github.com/ArduPilot/ardupilot/issues/11434), we should never allow arming in autonomous modes (Loiter, PosHold, Guided, etc) without a good position estimate even if the user sets ARMING_CHECK = 0 or does a "forced arming". It's not safe because the vehicle cannot possibly control it's position nor will it be accepting input from the pilot (because it's an autonomous mode).

bluuu81, 2019-05-29:
Can't agree. If user set ARMING_CHECK to 0 they should know what he/she can do.
It is very useful for "desk" test.

Pedals2Paddles, 2019-05-29:
My first thought was that as well.  But reality if if you need to arm for testing without pre-arm checks, do it in stabilize mode. I cannot think of any good, reasonable use case to arm an aircraft in those other modes without the required supporting data.  I can however, think of lots of operator error that can result from being allowed to arm in a state that cannot possibly work correctly.  The latter IMO outweighs any small edge case.

peterbarker, 2019-05-30:
On Wed, 29 May 2019, bluuu81 wrote:

> Can't agree. If user set ARMING_CHECK to 0 they should know what he/she can do.

Amusingly, Randy and I are thinking that force-arming via mavlink should 
be equivalent to setting your ARMING_CHECK to zero.  At the moment it's 
stronger than that so you can bypass more than you can with setting the 
ARMING_CHECK to zero.

> It is very useful for "desk" test.

How is it a useful test?  What will you learn that arming in (e.g.) 
STABILIZE won't also tell you?  "Whether EKF failsafe triggers" isn't a 
reasonable response :-)

rmackay9, 2019-12-16:
This is resolved in master so closing

closed 2019-12-16 by rmackay9