ArduPilot/ardupilot #11440: ROS: Using GPS/IMU from Ardupilot in Cartographer

lukehutch, 2019-05-29:
### Feature request

**Is your feature request related to a problem? Please describe.**
I found the following description of using Google Cartographer's SLAM functionality to replace GPS, compass and IMU for indoor navigation:

http://ardupilot.org/dev/docs/ros-cartographer-slam.html

However, I want to do the opposite: I want to use Ardupilot's drivers to access GPS, compass, accelerometer and barometer information from a Here+ v2 GNSS device, and feed those values into Cartographer in order to improve the absolute positioning of nodes in the pose graph. For example, I need the following drivers to access the GPS and barometer, but I need to access these values from Cartographer:

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_UBLOX.cpp
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro_MS5611.cpp

(similarly for the inertial sensor, rate gyro, and compass -- I haven't figured out which chips those are yet on the Here+ v2).

Ideally I would like to run Ardupilot on Linux, and use a simple USB UART dongle to directly connect to the Here+ v2 sensor, and find some way to connect Ardupilot to Cartographer (even if it just means writing out the Ardupilot pose information to disk with accurate timestamps, and fixing the SLAM global snapping later in postprocessing, although realtime integration would be better).

I see I can run Ardupilot on Linux as SITL, but that seems designed for simulation, not for integrating real sensors with other software running on Linux (or on ROS on Linux, via Docker).

**Describe the solution you'd like**
There should be an easy way to use the Ardupilot drivers directly on ROS or Linux to read sensor data.

**Describe alternatives you've considered**
See above. I'm really not sure of the right way to proceed, but I haven't found a solution yet.

**Platform**
[X] All (sensors)

**Additional context**
Some info on the Here+ v2: https://www.getfpv.com/here2-gnss-gps-module.html
It plugs straight into the Pixhawk 2.1.

khancyr, 2019-05-29:
Hello,
for user question please use https://discuss.ardupilot.org/ 
About some answer, if you want to only use a here gps ... then just connect it on your linux board. ROS already have a node for ublox and barometer ... There is no need of ardupilot in that case. Otherwise, use mavros and will you have all correct ROS topic published

closed 2019-05-29 by khancyr

lukehutch, 2019-05-29:
Thanks for the explanation, that really helps. (I asked already on the forum, and didn't get an answer.)