ArduPilot/ardupilot #11449: EKF origin can be inconsistent between EKF lanes

tridge, 2019-05-30:
The EKF origin is stored separately for each lane. This can cause a few issues:
 - the origins could be different if we get a sharp GPS change between timesteps of the EKF update so the two lanes get different origins
 - the mavlink set origin command only sets the primary, allowing for the two to be very different
This becomes an issue on an EKF lane change, as AP_InertialNav uses the EKF origin for copter and quadplane
thanks to @WickedShell for noticing this
ping @priseborough

tridge, 2019-06-03:
check origin drift with multiple cores

rmackay9, 2020-04-29:
@tridge, I think we've resolved this now, what do you think?

IamPete1, 2021-09-12:
@rmackay9 @tridge Any update? Fairly sure we can close?

tridge, 2021-09-13:
yes, origin is now guaranteed the same (same variable) between EK3 lanes.
can be inconsistent between EKF2 lanes

tridge, 2021-09-13:
the final fix will be part of the work @peterbarker is doing on AHRS to move public origin up to AHRS level