ArduPilot/ardupilot #11454: sim_vehicle.py does not exist

PouriaMzn, 2019-05-31:
#### Issue details
I'm trying to set up SITL on Windows 10, i installed Cygwin as is described [here](http://ardupilot.org/dev/docs/building-setup-windows-cygwin.html#building-setup-windows-cygwin), then installed  MAVProxy, and cloned Ardupilot source files to the Cygwin filesystem. But when i try to execute :

```
cd ~/ardupilot/ArduCopter
sim_vehicle.py --map --console
```

I get this massage: `-bash: sim_vehicle.py: command not found`
I checked the directory and there's no `sim_vehicle.py` file in it to run. Can someone help to figure this out, I don't know what i'm doing wrong.

rajat2004, 2019-05-31:
Hi,
I haven't actually used Ardupilot on Windows before, but this happens on Linux when the `Tools/autotest` directory hasn't been added to the PATH, see this page: http://ardupilot.org/dev/docs/building-setup-linux.html#add-some-directories-to-your-search-path-facultative
(`sim_vehicle.py` is present in the `Tools/autotest` directory)
So I'm not sure if there's something like the PATH in Cygwin, you can try this out, or else, directly use the path of sim_vehicle.py 
Something like ` ../Tools/autotest/sim_vehicle.py --console --map`

Someone who has used Windows would be better able to help out, but I'm not sure how something like this would have gone unnoticed till now.

magicrub, 2019-05-31:
The problem is likely just that you don't have Tools/autotest in your path. There's a new script for Cygwin installaion that I don't think is on the wiki yet. Try running the [windows ps1 script](https://github.com/ArduPilot/ardupilot/tree/master/Tools/environment_install). If you're using Windows 10, I highly suggest you try [WSL](http://ardupilot.org/dev/docs/building-setup-windows10.html), it's what I use and works great. It's like a mini-virtualMachine on Windows 10 and runs just like Ubuntu CLI. Just run the Ubuntu install script!

closed 2019-05-31 by magicrub

PouriaMzn, 2019-05-31:
I solved the issue by adding the environment variable to the file `sim_vehicle.py` that is in `ardupilot\Tools\autotest\` directory, then i executed this code 
`export PATH=$PATH:$HOME/ardupilot/Tools/autotest` in Cygwin terminal and everything just worked fine.

khancyr, 2021-03-31:
You need to install pymavlink.
Please follow our instructions on the the wiki

magicrub, 2021-03-31:
@Rajeta-Sarkar I highly recommend using WSL instead of cygwin

UbhayAnand1, 2023-01-29:
@khancyr I used this link to Download MAVProxy for Windows](https://firmware.ardupilot.org/Tools/MAVProxy/MAVProxySetup-latest.exe). I still am facing the issue

khancyr, 2023-01-29:
I have no experience with cygwin... So I won't be much help.

Better use wsl.

Otherwise, use sim_vehicle.py and add --no-mavproxy and use your windows mavproxy to connect the sitl instance. But that isn't worth the trouble against using wsl directly

UbhayAnand1, 2023-01-29:
@khancyr I did ln -s /usr/bin/python3.7 /usr/bin/python
ln -s /usr/bin/pip3.7 /usr/bin/pip
pip install empy pyserial pymavlink    but it fails, and the error says something about lxml.

UbhayAnand1, 2023-01-29:
Ok, @khancyr thanks. Sorry i deleted the messages cause i had not refreshed the page and did not see your reply.... Actually i figured it out, now I have to learn SITL.... But would you still recommend me to go with WSL, or should I now continue with Cygwin. It took me 6 hrs just to finally finish it.

magicrub, 2023-01-29:
@UbhayAnand1 you can try doing pip install lxml and see what fails next, but I highly recommend using wsl instead.

UbhayAnand1, 2023-01-30:
the pip install lxml, was failing due to some wheel error. but I went to the bin, and deleted the pip and python files,and ran the commands to install pymavlink from the wiki again, and it worked... But I will also try to use wsl now, as it seems its better.