ArduPilot/ardupilot #11455: Firmware copter stop working on fmuv2

khancyr, 2019-05-31:
### Bug report

**Issue details**

This commit break fmuv2 : https://github.com/ArduPilot/ardupilot/pull/11319

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
tested on pixhawk 1
**Logs**
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2019-05-31:
We had someone on the forum who was also having problems with fmuv2 - although @peterbarker couldn't reproduce.

rmackay9, 2019-05-31:
I think this applies to master only and not Copter-3.6 but please tell me if I'm wrong (I'm in Tokyo this morning so can't immediately check myself)

khancyr, 2019-06-01:
Yep master only, all release seems fine. I didn't try other vehicles but it will failed to.
It need a waf clean to fail.

tridge, 2019-06-01:
@khancyr when you say "break" what do you mean? What are the exact symptoms.
I tested it on a Pixhawk1 here and it works fine.
Can you also please try changing the timing of it, specifically there are two loops in peripheral_power_enable(). If you increase the loop counts does it fix it?

khancyr, 2019-06-01:
Did you clean the build first ?
Here is a video of what I got : 
https://youtu.be/OaRhmEQchYU
I will try later to change the timing

tridge, 2019-06-01:
video link doesn't work.
Can you describe the error?

tridge, 2019-06-01:
it may also matter what bootloader you have, and how you are powering. Can you describe those please?

khancyr, 2019-06-02:
hello,
The error is that the boad keep disconnecting. Bootloader is accessible but the firmware doesn't start. On the pixhawk1 the orange led on FMU side keep blinking.
The bootloader is the one for pixhawk1.
The board is powered directly by usb, and got no periferical.

rmackay9, 2019-06-04:
I uploaded Copter master for fmuv2 to my regular desktop pixhawk1 and it worked.  It was a bit odd immediately after the upload in that after the board rebooted the gyro calibration failed but once rebooted once more it seemed to work.

khancyr, 2019-06-04:
@tridge @rmackay9 you got 3dr pixhawk 1, right? Mine is a chinese one's ... I guess I have another one from china, I will test

peterbarker, 2019-06-04:
On Tue, 4 Jun 2019, Pierre Kancir wrote:

> @tridge @rmackay9 you got 3dr pixhawk 1, right? Mine is a chinese one's ... I guess I have another one from china, I will test

I'm guessing minor hardware differences.... "nobody could possibly need 
these components, works fine without them...."

robustini, 2019-06-08:
I've exactly the same problem with a chinese Pixhawk that from the box looks exactly the same in my hands.
Flashed from the Planner with Plane 3.9.8 (fmuv2) no issue, flashed from the latest precompiled firmware on "Ardupilot Firmware" and it doesn't boot.
When I try to flash the .apj the Planner asks me if the board is a Solo, obviously I say no, flash and then zero activity, then the board is recognized with another COM and the firmware doesn't boot.
To solve this here it is enough to flash 3.9.8 Plane release and no problem.
And now?

tridge, 2019-06-08:
@khancyr please test this PR and see if it fixes it for you:
https://github.com/ArduPilot/ardupilot/pull/11538

closed 2019-06-09 by tridge

robustini, 2019-06-09:
I'm sorry @tridge , false positive, seemed to work but after a reboot from the same problem.

PR #11538 (2019-06-09): HAL_ChibiOS: move the power control changes to be only CubeBlack
This aims to fix #11455. I have been unable to reproduce the issue,
which leaves me thinking it is either a bootloader difference or a hw
difference between pixhawk1 clones and real pixhawk1 boards

@khancyr please test this PR and see if it fixes the issue for you
commit 51e18734 (2019-06-09): HAL_ChibiOS: move the power control changes to be only CubeBlack

This aims to fix #11455. I have been unable to reproduce the issue,
which leaves me thinking it is either a bootloader difference or a hw
difference between pixhawk1 clones and real pixhawk1 boards