ArduPilot/ardupilot #11460: SERIAL_OPTIONS option does not appear in search

manotroll, 2019-06-01:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

### Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

SERIAL_OPTIONS option does not appear in search

trying to follow these rules here to put the settings to use the parameters

SERIAL2_BAUD 57 
SERIAL2_OPTIONS 4 
SERIAL2_PROTOCOL 10

both SERIAL2_BAUD and SERIAL2_PROTOCOL appear the non-SERIAL2_OPTIONS

Naterater, 2019-06-02:
This is because the firmware you are using does not support the feature yet. It is only available in "master" or the "latest" build on firmware.ardupilot.org.

I have opened a documentation issue about it here: https://github.com/ArduPilot/ardupilot_wiki/issues/1763

manotroll, 2019-06-02:
hum ... so it's only available in beta?

IamPete1, 2019-06-02:
beta or latest depending on the vehicle

closed 2019-06-02 by IamPete1