ArduPilot/ardupilot #11467: Copter: support terrain following in Guided mode

rmackay9, 2019-06-02:
Terrain following works in manual modes (using lidar) and also some autonomous modes (Auto and RTL) but it is not supported in Guided mode.  To help with a developer looking to implement this what needs to be done is:

- modify mode_guided.cpp's set_destination() function (there are two but the one that needs modifying is the one that accepts a Vector3f so that it accepts a boolean terrain_alt parameter.  Then pass this boolean into the wp_nav->set_wp_destination() call within the function.
- modify GCS_MAVLink.cpp's MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT handler function so that they support the mav_frame parameter being MAV_FRAME_GLOBAL_TERRAIN_ALT or MAV_FRAME_GLOBAL_TERRAIN_ALT_INT.  if the mav frame is either of these two then when we call the mode_guided.set_destination() the new terrain_alt should be true.

We should probably also add support for Guided mode's set_destination_posvel to accept a terrain altitude although there are many fewer users of this controller.

khancyr, 2019-06-03:
Another nice addition on this would be to be able to activate or disable it.

rmackay9, 2020-03-25:
This is implemented in master now so closing.  I think this will be released with Copter-4.1.

closed 2020-03-25 by rmackay9

DaneSlattery, 2020-07-31:
Hi @rmackay9 

Could you please link the commit where this change was made?

Does this also extend to users sending direct attitude setpoints in guided mode? I know terrain following works in guided_nopgs, so I'd think one could just borrow the code from there for handling attitude setpoints in guided mode.

Thanks