ArduPilot/ardupilot #11479: External AHRS integration into EKF

Kiwa21, 2019-06-03:
### Feature request

**Is your feature request related to a problem? Please describe.**
Attitude estimation with "cheap" sensors found in most of the market autopilots might sometimes be a trouble for high end vehicle integration. For example, compasses are often an issue and a bad heading will result in poor flight performance.

**Describe the solution you'd like**
I want to be able to at least use an external AHRS to provide euler angles (pitch, roll, yaw) / quaterions to the Ardupilot EKF.
Using an INS would be interesting as well but I'm not sure how it can cope with the others sensors usable in the EKF (rangefinder, beacon, ...).

**Describe alternatives you've considered**
There is already one attempt to use an external high end INS here: 
**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

amilcarlucas, 2021-01-05:
Master has this now

closed 2021-01-05 by amilcarlucas

Tdoe4321, 2021-02-02:
> Master has this now

Is there a reference as to how to do this? I am looking at trying to integrate a beefy IMU into an ardusub system.

rmackay9, 2021-02-02:
@Tdoe4321,

We don't have any wiki docs yet but when we do they should appear here: https://ardupilot.org/copter/docs/common-master-features.html

The connection with the external ahrs system needs to be using a serial cable and the relevant parameters start with EAHRS_ in case that helps.

tridge, 2021-05-10:
@Tdoe4321 the code is here: https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_ExternalAHRS
it now supports adding new backends. Currently UART only, but CAN is planned

lebarsfa, 2021-11-03:
FYI, I just made a ROS node https://github.com/ENSTABretagneRobotics/ardupilot2ros that should be able to convert `/imu/data` and `/navsat/fix` typical ROS messages (e.g. from AHRS/INS like SBG Ellipse and others) to the VectorNav protocol read by ArduPilot. 
In addition, it turns out that we can also use `SERVO_OUTPUT_RAW` MAVLink messages from ArduPilot to generate `/cmd_vel` typical ROS messages, so it could be a quick way to test/use ArduPilot ecosystem (without hardware modification if we use an SITL ArduPilot firmware) on ROS-based robots such as some from Clearpath Robotics and others.

Ryanf55, 2023-11-11:
I'd like to consider re-opening this issue. I do not believe the implementation done actually meets this design request:
`I want to be able to use an external AHRS to provide euler angles (pitch, roll, yaw) / quaterions to the Ardupilot EKF.`

The implementation right now replaces the ArduPilot EKF with the external one. [If the external](https://ardupilot.org/copter/docs/parameters.html#ahrs-ekf-type-use-navekf-kalman-filter-for-attitude-and-position-estimation) is an enum, not a bitfield. The EAHRS data is not combined with any other data, and if EAHRS fails, then it appears ArduPilot falls back to DCM.