ArduPilot/ardupilot #11486: Copter/Rover: if no polygon fence is defined, skip polygon fence pre-arm checks

rmackay9, 2019-06-04:
Sometime between Copter-3.5.7 and Copter-3.6 we changed the pre-arm checks for AC_Fence such that if a user set FENCE_TYPE = 4 (for polygon) or any other option that include the polygon fence that they must also then define the polygon or a pre-arm check failure, "PreArm: Polygon boundary invalid" appears.

I think we should revert the behaviour back to what it was for 3.5.7 meaning that if a user does *not* define a fence, then we treat this the same as having FENCE_TYPE != 4.  I.e. we skip the polygon fence check if no points have been defined.

Some may argue that if the user says they want a polygon fence that it makes sense to pop-up a pre-arm error if there is no polygon fence defined *however* the default FENCE_TYPE is "7" which includes the polygon fence.  Many/Most users do not want to use the polygon fence, it's just that the default includes it.  I think it would be easier for the user to simply set one parameter (FENCE_ENABLE = 1) to enable the fences rather than then hitting this pre-arm check and having to go back in and change the FENCE_TYPE.

Another solution would be to change the FENCE_TYPE's default to just Circle.. but changing that now could mean that after users upgrade, their polygon fence no longer works.

peterbarker, 2019-06-04:
155407956c1b224c6c8311ccb392febe2508a07c is the commit which added these additional sanity checks.

peterbarker, 2019-06-04:
I think the current semantics are preferable think they should be extended - if you say you have an altitude FENCE_TYPE  then it must be >= 0, if you have a circular fence you must have a radius etc etc.  This is erring on the side of caution and permits the user to say they want to always fly with a polyfence.

rmackay9, 2019-06-04:
@peterbarker, I'm starting to think you're right actually.. I don't like how annoying it is for users to set it up.. but ..

IamPete1, 2021-08-29:
I think current behavior is correct, it prevents flying without a fence if you thought you have one. 

So maybe this can be closed?

IamPete1, 2021-08-29:
Tested this, turns out this is what it does currently. Sort of, it will fail if you only have polygon selected and no polygon.

IamPete1, 2021-09-11:
@rmackay9 Currently we do skip the polygon check is there is no fence defined, unless polygon is the only type of fence setup. What do you think the behavior should be? We could skip in that check too, but at that point you might as well just turn off fence.

peterbarker, 2024-08-03:
This is working as Randy wants it to work, so closing it.

You can have polygon fence enabled as a fence type, but that doesn't mean you need to have any.

closed 2024-08-03 by peterbarker