ArduPilot/ardupilot #11490: Plane: Elevon jitter on the ground when throttle up, but only in FBWA and with throttle disarmed

cheintz, 2019-06-04:
### Bug report

With my plane on the ground and safety-switch turned, on, I see elevon jitter under the following (combined) conditions:

Mode: FBWA
Throttle: disarmed
Throttle stick: Above 0%

Changing any of those makes the jitter go away. I have the ESC unplugged for this log, though the symptoms are the same with it plugged in.

The log includes various sequences of control movements and mode changes that demonstrate the problem

**Version**
3.9.8

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying Wing

**Hardware type**
PX4

**Logs**

[2 12-31-1979 7-00-00 PM.zip](https://github.com/ArduPilot/ardupilot/files/3250978/2.12-31-1979.7-00-00.PM.zip)

magicrub, 2019-06-24:
Thanks for the log, I'll take a look

tridge, 2019-08-26:
this is fixed now in master
please re-open this issue if you can still produce on master

closed 2019-08-26 by tridge