ArduPilot/ardupilot #11491: Propellers are moving when trying to download 500 logfiles through UART

jeroentaverne, 2019-06-04:
### Bug report

**Issue details**
While trying to download a lot of logfiles (500 in my case) through UART in stead of USB, the propellers start moving. After checking the signals to the speed controllers by oscilloscope, they seem to go to maximum for a short time.

**Version**
3.6.7

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Octa

**Hardware type**
Pixhawk 1

**Logs**
NA

jeroentaverne, 2019-06-04:
By changing MAX_LOG_FILES to 300 in DataFlash_File.cpp the problem is gone.

peterbarker, 2019-06-06:
Scope traces, please?

jeroentaverne, 2019-06-06:
Find them attached. I also noticed that during logfile transfers the oneshot starts jittering a lot. So the STM32F4 looptime is affected. I think at some point the oneshot output totaly stops and a failsafe (in the STMF32F1) might be activated which generates the 150usec pulse.
[image]
[image]

WickedShell, 2019-06-06:
@jeroentaverne What pin on the autopilot is this? With a pixhawk 1 it's important to distinguish between main and aux outputs. (A param file may also help someone try and reproduce this)

jeroentaverne, 2019-06-06:
It's M1 pin of Pixhawk1, not AUX. So it's an output of the STM32F1 IO processor. I have tried to find out what this processor is doing when the communication with the STM32F4 is lost for some time. But I can't find the recent source code.
Parameter file can be found here: https://we.tl/t-9XExx95H6k

tridge, 2019-06-10:
you're running 3.6.7 copter which uses nuttx IO fw. Can you please try master so you can compare with ChibiOS IO firmware? Note that 3.6.7 uses NuttX IO firmware even with ChibiOS on fmu

jeroentaverne, 2019-06-12:
Thanks, I will try. Which stable version has Chibios IO firmware inside?

jeroentaverne, 2019-06-14:
I have tried the latest 3.6.9 and also modified 3.6.7 to use fmuv2_IO_ChibiOS.bin. In both cases the oneshot125 outputs are not triggering motor rotation anymore and stay well under 126 usec, so that's good. But the signal to the motors is just stopped completely during log list download and then the ESCs starts generating alarm sounds. It really seems that SD card access is stopping the control loop. I thought that the control loop had the highest priority in the RTOS?

jeroentaverne, 2019-06-14:
By modifying the function DataFlash_File::get_num_logs() so it directly returns 500 without doing a lot of SD card access, the alarm sounds are gone, but there is still a lot of jitter in the periods between the oneshot125 signals. It looks like SD card access isn't interrupted at all by the control loop.

rmackay9, 2019-06-18:
on the dev call we discussed that the best solution was to get the mavlink message processing moved to a new thread.

jeroentaverne, 2019-06-18:
So you think the issue is caused by Mavlink and not related to SD card access? Is it possible to give me some guidance how to implement this new thread? Thanks. :-)

peterbarker, 2019-06-18:
On Mon, 17 Jun 2019, jeroentaverne wrote:

> So you think the issue is caused by Mavlink and not related to SD card access? Is it possible to give me some guidance how to implement this new thread?
> Thanks. :-)

The current theory is that interrupting the main thread do do file system 
operations is causing the problems.

It's a great theory.

It would be nice to verify it before we dive head-long into the 
gcs-on-a-thread idea - that will be fun, but involved and tricky :-)

If you could hack up some function somewhere to just pause the main 
thread for some (gradually increasing?) number of milliseconds and see if 
that replicates the problem - that would be handy.

jeroentaverne, 2019-08-22:
Is this problem somehow solved in 3.6.9 or 3.6.10. I noticed some changes to the source code involved in SD card access.

rmackay9, 2019-08-23:
@jeroentaverne, we haven't looked into this report much so I don't think we know if it is resolved in 3.6.9 or 10.  I suspect is hasn't been resolved.

jeroentaverne, 2021-06-07:
Has this been resolved? I see same problem with 4.0.3

rmackay9, 2021-06-08:
@jeroentaverne,

I still think it has not been investigated or resolved.  Now is perhaps a good time to resolve it as part of the 4.1 release though.  If it can be reproduced with 4.1 then I can pretty easily add it to the 4.1 issues list.

As a work around I wonder if setting MOT_SAFE_DISARM = 1 will resolve the problem.