ArduPilot/ardupilot #11494: Update nav target if home position changes while in RTL

AndKe, 2019-06-04:
### Feature request
I have just made, and tested this module at sea:
https://github.com/ArduPilot/MAVProxy/pull/633

Current behavior:
If an RTL is trigged, the vehicle will navigate to the RTL position at the time of switching to RTL.
No incoming MAV_CMD_DO_SET_HOME will affect the navigation target,

This is bad if the home is moving at some speed, or the travel time to home (moving ship) is significant.

**Describe the solution you'd like**
If home position is updated during RTL, the navigation should be updated to go to that, instead of the position at the time RTL was engaged.

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[x ] Plane
[x ] Rover
[x ] Submarine

Tested with AC 3.6.7 and AC3.7-dev

VDLJu, 2019-11-25:
Hi,

Please see following issue. This might be related to RTL behavior when home position is adjusted.
https://github.com/ArduPilot/ardupilot/issues/12888

IamPete1, 2021-09-12:
We have https://github.com/ArduPilot/ardupilot/issues/527 for plane.