ArduPilot/ardupilot #11505: Do_Set_Cam_Trigg_Dist command, wrong distance between photos

buckker, 2019-06-05:
I posted this issue a few month ago here: https://discuss.ardupilot.org/t/wrong-distance-between-photos-do-set-cam-trigg-dist/40596

I still have no solution for this behaviour. 

Sometimes I thought the bug is flying speed related. Lower flying speed=>higher number of trigger commands. But this evening I reduce the flying speed from 6.75m/s to 6m/s and the trigger commands are lower that with higher flying speed.
So, there must be somewhere a bug. 

I use Arducopter 3.6.8

Naterater, 2019-06-12:
Provide logs. Are you considering "triggered" photos, or just actual photos? Sometimes it's a camera issue, not an autopilot issue. Do you have shutter feedback?

buckker, 2019-06-13:
To be more precise, I don't use a photo camera. I use the signal to controll a servo and drop some small balls into the field for an agriculture project. I double checked if mission planner match the effectively dropped ball count. It match perfectly (Mission Planner last photo No. 127, I filled 130 balls to the container, after the flight 3 pcs were left)

IamPete1, 2021-08-29:
I think this is just a issue with the difference between the mission distance and the copter actually flys, we cut the corners slightly so you end up flying a slightly shorter distance and thus end up with a few less triggers. The faster you fly the more the corners are cut so the shorter distance you fly and end with a few less triggers. 

You could improve this slightly by setting a short delay at each way-point, this will force the copter to got to it and not cut the corner. 

Or you could have a waypoint for each location and use a `DO_SET_RELAY` for the trigger.

closed 2021-08-29 by IamPete1