ArduPilot/ardupilot #11519: Copter-3.5.x firmwares are missing from firmware.ardupilot.org

rmackay9, 2019-06-06:
We seem to have 3.2.1, 3.4.x, 3.6.x but no 3.5.x firmwares.
http://firmware.ardupilot.org/Copter/

tridge, 2019-06-08:
we don't have them in the archives. We could recreate them if you think users will want them

rmackay9, 2019-06-08:
I'm pretty sure that the mission planner has them, definitely users would like the 3.5.x versions.. or at least 3.5.7 (the last one)..

rmackay9, 2019-06-18:
Here are some links provided by @meee1 with the locations of the binaries for some of the 3.5.x releases

3.5.7: https://github.com/ArduPilot/binary/commit/6a11d84faeba5640395f404bb1a97c24f8e733ed
3.5.5: https://github.com/ArduPilot/binary/commit/ff0b633bcad2e5f2d5e992cce6381a66bdcc8180
3.5.3: https://github.com/ArduPilot/binary/commit/4f79f1b4d92b7cf52933377a3f1d5d8dbc432cab 
3.5.2: https://github.com/ArduPilot/binary/commit/d219f38debaccb825f6cebe108678bcea312782c

rmackay9, 2019-07-06:
It seems the binaries for Rover aren't quite right either... https://github.com/ArduPilot/ardupilot/issues/11732