ArduPilot/ardupilot #11534: AC_Avoid/AP_OAPathPlanning: consolidate on Vector2f instead of Location

rmackay9, 2019-06-08:
As mentioned in the review comments from @OXINARF on this PR https://github.com/ArduPilot/ardupilot/pull/11427, the AP_OAPathPlanning library (especially the BendyRuler class) uses a mixture of Location (i.e. Lat, Lon) and Vector2f (i.e. offsets from the EKF origin) to specify the absolute location of objects.  We should consolidate AP_OAPathPlanning (and AP_Proximity) to always use Vector2f/Vector3f - i.e. we should always use offsets from the EKF origin

peterbarker, 2019-06-08:
This may make integration with Plane problematic.

rmackay9, 2019-06-09:
@peterbarker, I think it'll be OK, we can expose a method for Plane that provides the destination as a Location.

WickedShell, 2019-06-10:
I think that's actually the wrong direction to expose it for plane, for two reasons:

1. Plane actually does benefit from enabling the EKF origin drifting to compensate for baro drift if not using GPS as primary altitude.
2. More signifcantly planes can fly far enough away that using Vector2f is a less accurate coordinate format then a Location. You could make the argument that it's precise enough, but a plane will easily start to fly far enough that the Vector2f version will be truncating the coordinates down.

rmackay9, 2019-06-10:
@WickedShell, good points but two counter arguments:

- this is 2D for now so moving the ekf origin up and down will be OK
- we can (and should) change the Vector2f position offset from the EKF origin to be in meters instead of cm.  If we do this then it should be accurate out to many hundreds of km.
- efficiency is key and working in Lat/Lon adds an overhead of constantly changing back to the equivalent of Vector2f for math operations (i.e. angles, offsets).

WickedShell, 2019-06-11:
> this is 2D for now so moving the ekf origin up and down will be OK

We may need to be enabling a 3D drift in the future for long distance flight anyways, see the later point that swapping to meters doesn't fix anything. And also realistically plane will need to transition to a much more 3D version of the algorithm I expect.

> we can (and should) change the Vector2f position offset from the EKF origin to be in meters instead of cm. If we do this then it should be accurate out to many hundreds of km.

Doesn't matter, in both cases anything past 100km will be better represented in Location then Vector2f. They have an identical scaling problem, and people are already trying to plan missions that fly that far (or further)

> efficiency is key and working in Lat/Lon adds an overhead of constantly changing back to the equivalent of Vector2f for math operations (i.e. angles, offsets).

You don't actually have to swap to Vector2f for those operations, there is the math to do that directly in lat/long (I'm not sure if we have all of that imported to ArduPilot or not at this point or not).

rmackay9, 2019-06-12:
OK, we've discussed on mumble and I'm at least partially convinced.  At least it's not clear cut that we should move to Vector2f.  Doing geometry/math operations with Lat/Lon points seems wasteful and slightly messy but perhaps that's not a strong enough argument.

closed 2019-06-12 by rmackay9