ArduPilot/ardupilot #11542: Rover: populate PIDA.Act and PIDS.Act dataflash log values

rmackay9, 2019-06-09:
Rover logs the outputs of the Speed controller as PIDA and the Turn Rate controller as PIDS.  For both, the "Act" field is always left at zero.  We should populate these values with the actual speed or turn rate using the AC_PID object's set_actual_rate() method.

- For the speed controller we should [add the line right here in AR_AttitudeControl.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/APM_Control/AR_AttitudeControl.cpp#L469).
- for the turn-rate controller we should [add the line right here in AR_AttitudeControl.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/APM_Control/AR_AttitudeControl.cpp#L364).

vishal-raj15, 2019-12-09:
hello @rmackay9  i am interested in working in this issue .
but i dont have any idea about solving this issue can help me .

rishabsingh3003, 2020-04-06:
I believe this issue was fixed somehow down the line...

[image]

rmackay9, 2020-04-07:
@rishabsingh3003, awesome, thanks!

closed 2020-04-07 by rmackay9