ArduPilot/ardupilot #11544: AP_Proximity: add "memory" of locations to allow removal of Rover's Dodge avoidance

rmackay9, 2019-06-10:
Related to this Issue:https://github.com/ArduPilot/ardupilot/issues/11534 and this PR: https://github.com/ArduPilot/ardupilot/pull/11427, we should do two things to improve AP_Proximity for use in object avoidance especially in Rover (although it applies to Copter as well):

- the copy_locations() method should provide earth-frame positions as an offset from the EKF origin instead of using Locations.  This will be more efficient both in terms of space (Vector2f is smaller than Location) and CPU (AP_OAPathPlanning converts them to Vector2f before using them)
- store a short history (perhaps 3 to 5 seconds worth?) of positions so that 1D lidars can be used for BendyRuler avoidance allowing Rover's "Dodge" method of avoidance to be removed.  If BendyRuler is paired with a 1D (forward facing) lidar, it will momentarily turn away from an obstacle ahead of the vehicle, but if this turn means the lidar no longer sees the obstacle then then a moment later (0.1second later) the vehicle will turn back again.  This will likely repeat over and over and the vehicle will come too close to the object.

magicrub, 2019-06-12:
I signed this to myself. Thanks for the details. I'll work on it on my flight to China next week

rmackay9, 2019-06-13:
@magicrub, OK great.  By the way, I think we should probably create a new class for this - inside the AP_Proximity folder but a new class.  The functionality is more important than the structure though in any case.. we can always fix the structure later.

rmackay9, 2019-08-03:
This issue has been completed now so closing.  We still need to remove Dodge but I'll raise a new issue for that.

closed 2019-08-03 by rmackay9