ArduPilot/ardupilot #11552: Copter: use terrain data for path planning (aka obstacle avoidance)

rmackay9, 2019-06-11:
ArduPilot Copter/Plane supports terrain altitude based missions ([wiki](http://ardupilot.org/copter/docs/terrain-following.html)).  This existing feature is good for mapping missions where the user is interested in maintaining a specific altitude above the terrain.

Somewhat separately some users want their vehicle to fly to a specific place and avoid terrain along the way (i.e. if there is a mountain between "home" and the first waypoint.  In these cases the user really wants terrain based avoidance and in these cases the vehicle should fly the shortest path to the waypoint while staying **at least** X meters from the terrain.

The difference can be see in the image below.  The red-ish line show the terrain based mission where the vehicle follows the contours of the land.  The dotted blue line shows the path that a terrain avoidance mission would follow.
[image]

apryamostanov, 2019-06-11:
Hi @rmackay9 ,

Thanks for creating this Enhancement issue.

I have a question:

What will happen if there is a mountain as shown below, on a vehicle path, in the context of this enhancement?

[image]

rmackay9, 2019-06-11:
@apryamostanov, yes, so definitely it should climb over that mountain.. so the shortest path would be above that mountain and then mostly straight across to the right..

apryamostanov, 2019-06-11:
In this case, this seems completely fine to me.
In my scenario I prefer using the dotted line - to record the video in mountains.

Just for information - adding same feature (to avoid mountain) into the terrain-based mission - it will be more difficult right? I.e. as an option for this mission configuration.
I don't opt for it, just asking for interest.

ddomit, 2023-09-22:
Hello Guys,

Its almost 4 years later now, any news on this feature? Can we help in any way?

Thanks

Ryanf55, 2024-08-24:
> Hello Guys,
> 
> Its almost 4 years later now, any news on this feature? Can we help in any way?
> 
> Thanks

Work with me getting this working in ROS. We need to get the AP terrain database available in grid_map_geo. 
https://github.com/ethz-asl/grid_map_geo/pull/38

Then we need to write a simple planner. 

Although we could do it in on the microcontroller, it's probably a lot quicker to do it offboard.