ArduPilot/ardupilot #11570: Adjust speed in relation to range of object detected

zurpv1, 2019-06-12:
### Feature request

**Is your feature request related to a problem? Please describe.**
No. 

**Describe the solution you'd like**
Add feature.

**Describe alternatives you've considered**
Using my old source code on the new rover via an Arduino Mega. Then trying to use MAVLink to connect to Mission Planner. However, this is probably beyond my knowledge at this point.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Additional context**
Is there a way to adjust the speed of the rover based on the distance an object is detected? If I missed it, I apologize.

For example: If no obstacle is detected, then 100% speed. If an obstacle is detected between 300-400cm, then 50% speed. I used such an approach a few years ago and I would like to use it again with ardurover if possible. 

Below is how I implemented it back in 2015. Please be kind, I am not a coder and I am sure that there is a much cleaner way to do it.  :)

// SONAR Center Forward   
void sonarCenterForward() {   
if (((cm[sonarCenter] >= distanceMin) || (cm[sonarCenter] == 0)) && 
       ((cm[sonarLeft] > steeringMinBuffer) || (cm[sonarLeft] == 0)) && 
       ((cm[sonarRight] > steeringMinBuffer) || (cm[sonarRight] == 0))) 
        {
         lcdDisplay("Forward Min ");  
         curGear = gearFwd; 
         curSpeed = forwardMin; 
         esc.writeMicroseconds(curSpeed);              // Forward, 1st speed
         
         if (((cm[sonarCenter] >= distanceMid1) && (cm[sonarCenter] != distanceMin)) && gyroTiltMinX() || (cm[sonarCenter] == 0))
            {
             lcdDisplay("Forward Mid1 "); 
             curGear = gearFwd; 
             curSpeed = forwardMid1; 
             esc.writeMicroseconds(curSpeed);         // Forward, 2nd speed
            
            if (((cm[sonarCenter] >= distanceMid2) && (cm[sonarCenter] != distanceMid1)) && gyroTiltMinX() || (cm[sonarCenter] == 0))
               {
                lcdDisplay("Forward Mid2 "); 
                curGear = gearFwd; 
                curSpeed = forwardMid2; 
                esc.writeMicroseconds(curSpeed);      // Forward, 3rd speed
           
               if (((cm[sonarCenter] >= distanceMax) && (cm[sonarCenter] != distanceMid2)) && 
                     gyroTiltMinX() || (cm[sonarCenter] == 0))
                  {
                   lcdDisplay("Forward Max "); 
                   curGear = gearFwd; 
                   curSpeed = forwardMax; 
                   esc.writeMicroseconds(curSpeed);    // Forward, 4th speed
                  }     
               }
            } 
         }        
}

rmackay9, 2019-06-12:
@zurpv1,

Maybe you've already seen it but our [object avoidance wiki page is here](http://ardupilot.org/rover/docs/rover-object-avoidance.html).  The vehicle will slow down before hitting objects if it's in any mode except manual mode and has a good position estimate (i.e. GPS lock).

We have more advanced methods of object avoidance (or more accurately path planning) coming in Rover-3.6.  In particular [BendyRuler](https://www.youtube.com/watch?v=KuJLxMEcPnU) and [Dijkstra's](https://www.youtube.com/watch?v=JKxQGt6XW60).

I think I'll close this issue if that's OK.

closed 2019-06-12 by rmackay9

zurpv1, 2019-06-12:
@rmackay9 
Thank you for the info. I will look over the wiki again. I will be keeping an eye out for 3.6. 

I appreciate all of you all for all of the hard work.

Feel free to close this request.

Take care,