ArduPilot/ardupilot #11582: AC_Avoidance: Dijkstra's VisGraph should hold distances in meters

rmackay9, 2019-06-13:
Dijkstra's [VisGraph class should holds distances in meters](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AP_OADijkstra.h#L85).  It currently holds the distances in cm perhaps because this saves a small amount in computation but even so we should change this to be meters.

asucada, 2019-11-21:
@rmackay9 I would like to work on this issue. Please assign me to this one.

rmackay9, 2019-11-21:
@asucada, done, thanks!

hendjoshsr71, 2021-09-15:
@rmackay9 
I'd like to get this one closed and had my first PR to do this. However, I would need to revisit the PR to insure it still works. 

Any thoughts on still doing this? I could expand it to remove cm from the whole library instead? (and getting really ambitious NEU. )

rmackay9, 2021-09-15:
@hendjoshsr71,

Sure, as long as we test it thoroughly so that we're sure that it works..

omsheladia, 2022-01-21:
@rmackay9 is this issue being resolved yet? As it is showing open, I was interested in solving it. Let me know if I can fix it. Thanks.

hendjoshsr71, 2022-01-23:
This is not solved yet. But my first PR had it solved but between then and now a lot of code has gone into this area so it needs to be retested. 

I'm quite sure the linked PR of mine will not work without additional changes.

ShirshenduR, 2023-12-18:
#25797

Rahulpanwar8, 2025-01-20:
Hi, I am a new contributor and I would like to work on this issue. Could you please assign it to me? Thanks!

rmackay9, 2025-01-20:
Hi @Rahulpanwar8, thanks!  feel free to just start working on it and create a PR.  I don't think that anyone else is working on it at the moment.

ashishu23, 2025-03-04:
hi @rmackay9  I would like to work on this issue. If anyone is not working on this then can I continue?

rmackay9, 2025-03-05:
Hi @ashishu23, sure, go ahead.  best of luck!

SohamP-sys, 2025-10-15:
Hi @rmackay9 i am new to open source, and would like to contribute to this

rmackay9, 2025-10-15:
Hi @SohamP-sys, sure, go for it!