ArduPilot/ardupilot #11584: Tracker: rename AntennaTracker.cpp to Tracker.cpp

rmackay9, 2019-06-14:
Where possible we like to keep the implementations for a class in a .cpp file with the same name as the class name.  Based on this we should rename AntennaTracker.cpp to just Tracker.cpp.

This should be a non-functional change..

ayusdas2000, 2019-09-05:
can you please elaborate. Shall just make the class from the main file into a .cpp file.

rajat2004, 2019-09-05:
@ayusdas2000 First, thanks for your interest in contributing to Ardupilot!
Actually, this has already been done in https://github.com/ArduPilot/ardupilot/pull/12156, linking this issue in the PR was forgotten
You could try having a look at some of the other good first issues

@rmackay9 This issue can be closed I think

closed 2019-09-05 by peterbarker