ArduPilot/ardupilot #11587: Copter: Slow yaw response while flying grid

Hasi123, 2019-06-14:
### Bug report

**Issue details**

While flying a tight grid with WP_YAW_BEHAVIOR set to 1, the yaw response is delayed after reaching the second turnpoint in the grid. I would expect it to make a coordinated turn just like when reaching the first turnpoint.
I have attached a log and the first person kml, that shows the problem.

**Version**
V3.6.9

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Problem reproduced on Quad and Octo frames.

**Hardware type**
Pixhawk, Cube

**Logs**

[Logs.zip](https://github.com/ArduPilot/ardupilot/files/3289576/Logs.zip)

GlidLov, 2019-06-14:
Exact same problem here! I would like coordinated turns.
instead i get a "knife" shaped turn: first waypoint approached smoothly,  then a full stop at the other corner, then a full acceleration to the other side, while doing an aggressive rotation to forward direction at the same time. i use default Yaw direction, drone keeps its front to flight direction

rmackay9, 2019-06-17:
I'm afraid that it's been designed to work this way.  I think there may be an issue on the issues list to look at this behaviour again to see if it can be improved.  In any case, it's not really a "bug" (so I've updated the description a bit) but rather an "enhancement" in case it matters..

IamPete1, 2021-07-26:
I think S-Curves fixes this, so I'm going to close, please re-open if the issue remains.

closed 2021-07-26 by IamPete1