ArduPilot/ardupilot #11590: Pixhawk 2.1 "cube" and Holybro sik radio: sometimes won't start

GlidLov, 2019-06-14:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

On a properly configured, large quadcopter, equipped with two here2 gps units, a holybro 500mW telemery dongle and dual bricks power units, Flight controller won't start: no led, no motor outputs etc (not even everytime - sometime it starts, most often if i connect flight controller to my laptop through usb port cable). After disconnecting devices one-by-one, i discovered that disconnecting 433mhz telemetry module let the system start normally. I can even plug the telemetry connector while system is powered (!!) and it will work flawlessy! But if i plug it back, then power cycle, it won't start again. I previously used many of the same modules without problems, so i don't think it's a powe issue. I use Telem1 port which i know it gives enough power to telem module.

**Version**
3.7-dev
**Platform**
[ X] Copter
**Airframe type**
QUAD
**Hardware type**
Cube + Holybro telemetry module v3 500mW

OXINARF, 2019-06-14:
As written in the lines you didn't remove, user problems need to be posted in the forum (http://discuss.ardupilot.org), this issues list is for feature requests and confirmed bugs.

closed 2019-06-14 by OXINARF

proficnc, 2019-06-14:
Power that module externally