ArduPilot/ardupilot #11594: Soaring mode stalls plane in AUTO mode

M230, 2019-06-15:
#### Issue details
With soaring mode enabled, in AUTO mode, Plane will hold the nose up probably in an attempt to reach waypoint altitude even though the engine is off. This results in a stall (and in my case a spin). This behavior can also be seen in [this video](https://www.youtube.com/watch?v=NwAX3cPvMqw&t=433) though without the spin.

#### Version
Arduplane v3.9.7

#### Platform
[X] Plane

#### Airframe type
Ranger 2000 (Pusher plane)

#### Hardware type
Pixhawk '2.4.8'

#### Logs
This graph shows demanded pitch increasing and being held throughout the stall:
[image]

Possibly related: TECS speed inaccurate, GPS shows velocity falling until plane spins.
[image]

magicrub, 2019-06-15:
Thanks for the report. Can you post the log please?

M230, 2019-06-15:
Here it is
[log.zip](https://github.com/ArduPilot/ardupilot/files/3293622/log.zip)

magicrub, 2019-06-15:
@robustini @samuelctabor ping!

magicrub, 2019-06-15:
Closing. Thanks for the log. Your log and your params said you weren't using an airspeed sensor. 

Airspeed based stall prevention is not guaranteed without an airspeed sensor enabled, it has to rely on throttle. Soaring overrides throttle. Soaring is not recommended without an airspeed sensor. The first line in the wiki "[Setting up Soaring](http://ardupilot.org/plane/docs/soaring.html#setting-up-soaring)" says so. 

Also, GPS.sp is ground speed. TECS.sp is airspeed. TECS_SYNAIRSPEED=0 means the synthetic airspeed estimate is disabled. Without an airspeed sensor it's taking a guess at what the airspeed is. GroundSpeed and AirSpeed will only ever line up if there's exactly zero wind which is usually never the case. As you gain altitude the wind is totally different (faster) than on the ground.

closed 2019-06-15 by magicrub

robustini, 2019-06-16:
I think you're doing something wrong with your AP's parameterization.
I did a lot of tests even without the airspeed sensor and this stall problem never happened to me.
However you will surely get a better result by installing an arispeed sensor.

M230, 2019-06-16:
Thank you for the responses. I'll install an airspeed sensor soon.
Another thing from the same log that that I was wondering about is the extreme throttle oscillation in RTL mode when climbing. This happens just before the stall. I haven't encountered this before I used soaring mode. In takeoff mode the throttle worked fine. Wasn't sure if it's worth making a new issue for so I'll just ask here for now.
[image]

samuelctabor, 2019-06-16:
Hi @M230

Soaring doesn't do anything in RTL mode so it shouldn't be directly responsible. However, if you set TECS_SPDWEIGHT to 2.0 this may have caused the change. The TECS logic is quite different when an airspeed sensor is used, so I recommend seeing if installing one fixes the problem 👍