ArduPilot/ardupilot #11601: AC_Avoidance: vehicles using Dijkstra's may pass too close to fence

rmackay9, 2019-06-17:
When Dijkstra's populates the visibility graphs with distances between safe "nodes" it only checks if the points do not cross the polygon fence.  This can mean that a safe path is created that comes within cm of the fence.  Ideally we should use the "inner fence" instead of the original fence when calculating "visibility" but this bumps into the problem that when we want to calculate the visibility/distances between the inner polygon fence points they will always intersect with the inner fence because they are on the inner fence.

Below shows an example situation where this can happen
[image]

magicrub, 2019-06-17:
We need to implement [TRAJECTORY_REPRESENTATION_WAYPOINTS](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L5063) or [TRAJECTORY_REPRESENTATION_BEZIER](TRAJECTORY_REPRESENTATION_BEZIER) so this can be visualized and communicated at run-time.

weihli, 2019-06-19:
@rmackay9  I'm doing a job to avoid the no fly zone(multi-polygons) while vehicle return to launch, but now I'm so glad that dijkstra algorithm had integrated into ardupilot.   In function create_polygon_fence_visgraph, the condition j<_polyfence_numpoints-1 in https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AP_OADijkstra.cpp#L235 ignore the last point of _polyfence_pts because polyfence_pts is "unclosed"

rmackay9, 2019-06-19:
@weihli, are you sure it's avoiding the last point?  Peter Barker recently changed our Polygon handling functions (that start with Polygon_) so they don't need the last redundant point.

We don't have stay-out zones in master (but it's coming soon I hope) ..

weihli, 2019-06-19:
@rmackay9  I don't think it's a redundant point
For example if the destination's point near by the last fence point, and the line from current to destination point breach the fence, it also need to calculate the distance between the last fence point and current point for update the visible graph

weihli, 2019-06-19:
@rmackay9  In function create_polygon_fence_with_margin, the boundary of polygon fence is "unclosed",  So _polyfence_pts of smaller polygon is also "unclosed" meaning the last point is not the same as the fisrt.

PS: Creating smaller polygon with margin, there's an extreme case that two edge may be overlapping

weihli, 2019-06-19:
@rmackay9  I check the AC_Fence library,  in the function get_boundary_points , num_points  equal to total fence point minus two point(one for return point, the other closing point), in https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AP_OADijkstra.cpp#L151,  we get an unclosed polygon fence, So the last point is not a redundant point

weihli, 2019-06-19:
@rmackay9 Yes, the inner fence is unclosed, but in creating visibility graph, the condition j<_polyfence_numpoints-1 ignore the last point of inner polygon.

The inner polygon may be self-intersect polygon when two edges patallel and margin big enough

rmackay9, 2019-06-23:
@weihli, yes, I think you are right!    I hope to create a Parthia week to fix this issue

rmackay9, 2019-06-24:
@weihli, I've [created a PR that I think fixes this issue](https://github.com/ArduPilot/ardupilot/pull/11648) (and a couple of related issues in the same file). Could you have a look to confirm you think this is a good fix?

weihli, 2019-06-25:
@rmackay9  Yes, you have fixed it. 
 I think we should add a method to check the polygon is a simple polygon not self-intersection polygon.

rmackay9, 2019-06-25:
@weihli, great, thanks for checking.  My view is that we should move to always using "unclosed" polygons at least for all levels above the storage level.  At the storage level we may need the final point to ensure the sender has sent us the right polygon but besides that it's just a waste of space I think.

weihli, 2019-06-25:
@rmackay9  Yes, It's a waste of space with saving 'closed' polygon.  It also can not ensure it's  a right polygon(simple polygon). Moreover, The inner polygon with margin may be a self-intersection polygon.
So i suggest adding a method to comfirm the polygon is a simple polygon but not a self-intersection polygon.
The sorage is not big enough to save complex polygon points, should we save it in microsd card.

rmackay9, 2019-06-26:
@weihli, I didn't realise that the polygon could be self-intersecting.  It doesn't seem to cause problems for regular avoidance but certainly the new OAPathPlanning won't work.  I will soon be added some health checks to object avoidance so the user is more aware when it fails so this will at least tell the user that OA is not working when a self-intersecting polygon is used

peterbarker, 2019-09-04:
@weihli you may be interested in https://github.com/ArduPilot/ardupilot/pull/12167/files - exclusion zones, amongst other things.