ArduPilot/ardupilot #11619: ChibiOS: intermittent no-link messages on mavproxy

arjun7965, 2019-06-18:
### Bug report

**Issue details**
We have a pixhawk cube (3.6.9/master branch) connected to a Raspberry pi compute model via USB. Mavproxy on the Rpi intermittently display "no link" messages. During this nothing can be read from the pixhawk like "battery_status" or "SYS_STATUS" for example but, a reboot command goes through and reboots the pixhawk.  

Has anyone else had this issue? 

**Version**
Copter-3.6.9 and master branch 

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_N/A_

**Hardware type**
_Cube_

OXINARF, 2019-06-20:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

closed 2019-06-20 by OXINARF

rozzin, 2019-06-20:
The subject should be something more like "MAVLink output on Cube USB-serial link unexpectedly stops, produces only EOF when reading from USB host".

It looks like a bug to me.

I know the way @arjun7965 wrote the subject, it sounded like it could just be a MAVProxy usage issue; but "no output at all, just EOF" on the USB-serial interface as seen by the host computer is clearly something else.

It really looks like the the link on the autopilot somehow becomes one-way-only (packets go in, they don't come out). If this was actually a serial link instead of the USB link, I could see how messing up the flow-control settings could cause something like this..., but I don't think there even are flow-control settings for the USB serial emulation, are there?

OXINARF, 2019-06-20:
> The subject should be something more like "MAVLink output on Cube USB-serial link unexpectedly stops, produces only EOF when reading from USB host".
>
> It looks like a bug to me.

@rozzin No, it looks like an issue. There is no evidence of a bug and it certainly isn't a **confirmed** bug.

I'm sorry but if we allow support here, the list would be at 10000 open issues instead of the (already unmanageable) 1200.