ArduPilot/ardupilot #11632: manifest.json incorrect generation for heli firmwares

DonLakeFlyer, 2019-06-21:
This causes QGC to not be able to load heli firmwares on all boards. I'm not sure if it is just "OFFICIAL" versions which are incorrect. But those are the only entries which QGC uses for now.

Here is an example of an incorrect entry:
```
        {
            "mav-type": "Copter", 
            "vehicletype": "Copter", 
            "mav-firmware-version-minor": "6", 
            "format": "hex", 
            "url": "http://firmware.ardupilot.org/Copter/stable/CubeBlack-heli/arducopter-heli.hex", 
            "mav-firmware-version-type": "OFFICIAL", 
            "mav-firmware-version-patch": "9", 
            "mav-autopilot": "ARDUPILOTMEGA", 
            "platform": "CubeBlack-heli", 
            "mav-firmware-version": "3.6.9", 
            "git-sha": "632be63f053d45ab584190586d4a028abca9a2e4", 
            "mav-firmware-version-major": "3", 
            "latest": 0
        }, 
```
For this one mav-type should be "HELICOPTER" and platform should be just "CubeBlack".

Not all are wrong, here is an example of a correct entry:
```
        {
            "mav-type": "HELICOPTER", 
            "vehicletype": "Copter", 
            "mav-firmware-version-minor": "6", 
            "format": "arducopter-heli", 
            "url": "http://firmware.ardupilot.org/Copter/stable/navio-heli/arducopter-heli", 
            "mav-firmware-version-type": "OFFICIAL", 
            "mav-firmware-version-patch": "9", 
            "mav-autopilot": "ARDUPILOTMEGA", 
            "platform": "navio", 
            "mav-firmware-version": "3.6.9", 
            "git-sha": "632be63f053d45ab584190586d4a028abca9a2e4", 
            "mav-firmware-version-major": "3", 
            "latest": 0
        }, 
```

DonLakeFlyer, 2019-06-21:
@tridge I also need a version of the manifest in regular zip format not gzip format so I can use the zip version in QGC (currentl downloading unzipped version). This is a quirk of the cross platform support I have available to me. I can't unzip gzip cross platform. But I can regular zip. In order to make it work I also need the size of the original file in a seperate file I can read. Something like manifest.json.size or something like that with just the size written to it.

peterbarker, 2019-06-22:
On Fri, 21 Jun 2019, Don Gagne wrote:

> Not all are wrong, here is an example of a correct entry:

This doesn't look right, actually.

>         {
>             "mav-type": "HELICOPTER",
>             "vehicletype": "Copter",
>             "mav-firmware-version-minor": "6",
>             "format": "arducopter-heli",

That makes no sense a a format.

DonLakeFlyer, 2019-06-23:
> That makes no sense a a format.

Yup, didn't notice that. I tried fixing the python script myself but it seems to run from some special build which needs a "binaries" directory as input which I couldn't figure out how to build.

peterbarker, 2019-06-24:
On Sun, 23 Jun 2019, Don Gagne wrote:

>       That makes no sense a a format.
> 
> Yup, didn't notice that. I tried fixing the python script myself but it
> seems to run from some special build which needs a "binaries" directory as
> input which I couldn't figure out how to build.

./Tools/scripts/build_binaries.py --tags=dirty

That will create you a binaries directory (in ../buildlogs)

DonLakeFlyer, 2019-06-26:
@peterbarker The other ```"format": "arducopter-heli"``` you found is caused by this entry in the firmware repository: "format": http://firmware.ardupilot.org/Copter/stable/navio-heli/.

[image]

There is a file in there ```arducopter-heli``` with no extension. Not sure what that is.

DonLakeFlyer, 2019-06-26:
Another problem I just found is that many of the entries are missing the boatloader and vid/pid entries.

DonLakeFlyer, 2019-06-26:
> Another problem I just found is that many of the entries are missing the boatloader and vid/pid entries.

Nope. Caused by a bad command line.

peterbarker, 2019-06-27:
> There is a file in there `arducopter-heli` with no extension. Not sure what that is.

It's an elf - being a linux command we don't tend to tack on any extension.

I've corrected the detection of the file type in this PR: https://github.com/ArduPilot/ardupilot/pull/11666

Many thanks for digging into this.

I'm really not sure how much information we should be adding into this format field - these are all container formats and don't really tell you what's inside.