ArduPilot/ardupilot #11636: Compass not detected on OmnibusF7V2.

ninja-zx11, 2019-06-22:
OmnibusF7V2 running latest` Arducopter 3.7.0-FIRMWARE_VERSION_TYPE_DEV`

1)Drotek GPS+HMC5883 combo Compass is not detected.Its wired to SCL/TX3 and SDA/RX3 pads of the omnibusF7V2.UART3 is shared with I2C . SCL/TX3 connected to PB10 and SDA/RX3 connected to PB11 pin of the MCU.I don't know how this works but in the hw def file,it says 

> PD9 USART3_RX USART3
PD8 USART3_TX USART3

Does it look right?

`
# hw definition file for processing by chibios_pins.py
# for OMNIBUSF7V2 hardware.
@ thanks to betaflight for pin information

# MCU class and specific type
MCU STM32F7xx STM32F745xx

define CONFIG_HAL_BOARD_SUBTYPE HAL_BOARD_SUBTYPE_CHIBIOS_OMNIBUSF7V2

# board ID for firmware load
APJ_BOARD_ID 121

# crystal frequency, setup to use external oscillator
OSCILLATOR_HZ 8000000

define STM32_LSECLK   32768U
define STM32_LSEDRV   (3U << 3U)

define STM32_PLLSRC     STM32_PLLSRC_HSE
define STM32_PLLM_VALUE 8
define STM32_PLLN_VALUE 432
define STM32_PLLP_VALUE 2
define STM32_PLLQ_VALUE 9

FLASH_SIZE_KB 1024

# reserve one sector for bootloader and 2 for storage
FLASH_RESERVE_START_KB 96

# board voltage
STM32_VDD 330U

# only one I2C bus
I2C_ORDER I2C2

# order of UARTs (and USB)
UART_ORDER OTG1 USART6 USART1 USART3

PA11 OTG_FS_DM OTG1
PA12 OTG_FS_DP OTG1

PA13 JTMS-SWDIO SWD
PA14 JTCK-SWCLK SWD

PE0 LED0 OUTPUT LOW
PD15 BUZZER OUTPUT

# ICM-20608 on SPI1
PA4 MPU6000_CS CS

# MPU6500 on SPI3
PA15 MPU6500_CS CS

# SPI1 for IMU and baro
PA5 SPI1_SCK SPI1
PA6 SPI1_MISO SPI1
PA7 SPI1_MOSI SPI1

# SPI2 for MAX7456 OSD
PB12 MAX7456_CS CS
PB13 SPI2_SCK SPI2
PB14 SPI2_MISO SPI2
PB15 SPI2_MOSI SPI2

PC10 SPI3_SCK SPI3
PC11 SPI3_MISO SPI3
PC12 SPI3_MOSI SPI3

# SPI4 for SDCard
PE4 SDCARD_CS CS
PE2 SPI4_SCK SPI4
PE5 SPI4_MISO SPI4
PE6 SPI4_MOSI SPI4

# optional I2C device2, shared with USART3
PB10 I2C2_SCL I2C2
PB11 I2C2_SDA I2C2

# baro is BMP280 on SPI1
PA1 BMP280_CS CS

# mag is on I2C if connected

PC3 BATT_VOLTAGE_SENS ADC1 SCALE(1)
PC2 BATT_CURRENT_SENS ADC1 SCALE(1)
 
PC5 RSSI_ADC ADC1

# USART1
PA10 USART1_RX USART1
PA9  USART1_TX USART1

# USART2 for RC input, RX only
PA3 TIM9_CH2 TIM9 RCININT FLOAT LOW

# USART3
PD9 USART3_RX USART3
PD8 USART3_TX USART3

# USART6
PC7 USART6_RX USART6
PC6 USART6_TX USART6

# UART7, RX only?
#PE7 UART7_RX UART7

PB0  TIM3_CH3 TIM3 PWM(1) GPIO(50)
PB1  TIM3_CH4 TIM3 PWM(2) GPIO(51)
PE9  TIM1_CH1 TIM1 PWM(3) GPIO(52)
PE11 TIM1_CH2 TIM1 PWM(4) GPIO(53)

define HAL_STORAGE_SIZE 16384
define STORAGE_FLASH_PAGE 1

# spi devices
SPIDEV bmp280  SPI1 DEVID1 BMP280_CS MODE3   1*MHZ  8*MHZ
SPIDEV mpu6000 SPI1 DEVID2 MPU6000_CS MODE3  1*MHZ  4*MHZ
SPIDEV mpu6500 SPI3 DEVID1 MPU6500_CS MODE3  1*MHZ  4*MHZ
SPIDEV sdcard  SPI4 DEVID1 SDCARD_CS MODE0 400*KHZ 25*MHZ
SPIDEV osd     SPI2 DEVID1 MAX7456_CS MODE0 10*MHZ 10*MHZ

define HAL_OS_FATFS_IO 1
define HAL_BOARD_LOG_DIRECTORY "/APM/LOGS"
define HAL_BOARD_TERRAIN_DIRECTORY "/APM/TERRAIN"

# no built-in compass, but probe the i2c bus for all possible
# external compass types
define ALLOW_ARM_NO_COMPASS
define HAL_COMPASS_DEFAULT HAL_COMPASS_NONE
define HAL_PROBE_EXTERNAL_I2C_COMPASSES
define HAL_I2C_INTERNAL_MASK 0
define HAL_COMPASS_AUTO_ROT_DEFAULT 2

define HAL_BARO_DEFAULT HAL_BARO_BMP280_SPI

define OSD_ENABLED ENABLED
define HAL_OSD_TYPE_DEFAULT 1
ROMFS_WILDCARD libraries/AP_OSD/fonts/font*.bin

# we are low on flash on this board
define HAL_MINIMIZE_FEATURES 1`

ninja-zx11, 2019-06-28:
Solved;Needed 2.2k pullups on I2C.

closed 2019-06-28 by ninja-zx11

st0rfrasarn, 2021-01-18:
Hi,

How did you manage the pullups? Is physical resistors you are talking about or internal? I have the same problem. Maybe you could explain how you did? 

Thanks! :)

st0rfrasarn, 2021-01-18:
Ok, i soldered 2.2 pullups to SDA/SCL. Compass is still sketchy. But i am talking about the internal compass. Is your internal compass working?

ninja-zx11, 2021-01-18:
Genuine HMC5883 compass is working fine. You may have bad clone compass chip as these days market is full of clones.

ninja-zx11, 2021-01-18:
> Ok, i soldered 2.2 pullups to SDA/SCL. Compass is still sketchy. But i am talking about the internal compass. Is your internal compass working?

There is no internal compass on OmnibusF7V2.

st0rfrasarn, 2021-01-18:
aha. thats why. i swear i read somewhere it has internal compass. ok problem solved. thanks!