ArduPilot/ardupilot #11646: On GPS glitch , quad copter is in RTL mode

developermastermind07, 2019-06-24:
**Issue details**
We are using pixhawk 4 for quad copter. When gps glitch happens then quad copter goes to RTL flight mode and one quad copter crashes. 

**Version**
3.6.8 onwards (tested on 3.6.8,3.6.9 amd current beta version)

**Platform**
 Copter

**Airframe type**
quad copter

**Hardware type**
Pixhawk

rmackay9, 2019-06-24:
@developermastermind07,

This is really a support issue that should go in the forums until it is a confirmed issue.

I've had a quick look at our EKF failsafe (that should trigger during a GPS Glitch) and the only possible actions are AltHold and Land ([see here](https://github.com/ArduPilot/ardupilot/blob/Copter-3.6/ArduCopter/ekf_check.cpp#L151)).

Do you have a log file?

closed 2019-06-24 by rmackay9

developermastermind07, 2019-06-25:
> @developermastermind07,
>
> This is really a support issue that should go in the forums until it is a confirmed issue.
>
> I've had a quick look at our EKF failsafe (that should trigger during a GPS [...]

Yes ... PFA attachment for logs...
[2019-06-25 16-17-48.zip](https://github.com/ArduPilot/ardupilot/files/3325168/2019-06-25.16-17-48.zip)

rmackay9, 2019-06-26:
@developermastermind07,

It looks like the RTL was triggered by a fence breach (not an EKF failsafe).

The GPS's satellite count and HDOP look very bad so I'm a little surprised that the EKF failsafe didn't trigger but then I seem some other very odd things about this flight.  For example one motor (motor 3) is pegged at 100% while others are at zero.  I suspect this vehicle wasn't actually flying.