ArduPilot/ardupilot #11650: Reverse possition assist (motor trust) not working on quadplane while it is in any Copter-mode.

pompecukor, 2019-06-24:
### Bug report

According to the Wiki reverse assist should work, just as forward. This is not the case.

While forward assist works fine. We have tested and found that motor assist backward does not function in quad mode. No matter how we set it up. Also is not implemented in SITL either.

I can imagine that very few would ever try this as it need reverse trust capable ESC on the pusher motor of a traditional VTOL fixed-wing.

**Version**
Lastest master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Any (none-tilt rotor) VTOL airplane

Revers trust is setup correctly on the plane. It functions during fixed wing flights. However when in Quadmode, the only time we get assist is if pushing forward. Motors never spin reverse no matter how we try to fly backwards.

We tried "Use_reverse" = "always" even. No matter how much gain we put on Q_WVANE_GAIN.

[image]

IamPete1, 2021-09-12:
Yep, its not allowed

https://github.com/ArduPilot/ardupilot/blob/659286ee4fe3895e164e2110f23be715e60e941c/ArduPlane/quadplane.cpp#L3230-L3232

The question is should we allow it or update the wiki?

peterbarker, 2024-08-20:
Marked for DevCallEU to decide whether we should update the Wiki or the code.

tridge, 2024-08-28:
I think this should be able to work, although usually you would weathervane nose into the wind so this would not normally be needed.

tridge, 2024-08-28:
@Hwurzburg let's fix wiki now to say this doesn't work (or remove the text saying it does work) then if/when we fix it we can update again

Hwurzburg, 2024-08-29:
done

closed 2024-08-29 by Hwurzburg