ArduPilot/ardupilot #11674: All motors stop mid flight after RTL request, autopilot still functioning, aircraft crashes

Am5x86, 2019-06-27:
### Bug report

**Issue details**
Aircraft to be used to survey field of potatoes (weekly surveys, this being the fifth week). Aircraft taken off in stabilise mode to ~5m altitude, before alt hold and loiter flight modes were checked. Then auto enabled to start pre planned mission. A few seconds afterwards I realised the lens cap was on the thermal camera and initiated an RTL. Approx. 1 second after that, all motors switched off (eerie silence) and the aircraft crashed from approx. 16 m altitude.

On checking the logs I can see that the autopilot was still functioning and trying to give motor commands to fix the situation. However almost all current draw had ceased (see figure below) as all of the motors had stopped at the same time. The autopilot is powered via the PDB so it is not battery failure (both were OK) or PDB failure or a brownout. I cannot see in the logs any command or error that would stop all of the motors at the same time. Can anyone take a look at these logs and identify what may have occurred?

[image]

The aircraft is using an older version of arducopter (3.5.5) as it has been stable and performing well since March last year, therefore no reason to update it. The only thing I am doing differently this year is setting a heading hold command, but it has flown successfully multiple times since adopting this practise for this year’s surveys.

**Version**
Arducopter 3.5.5

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Octocopter (Flat)

**Hardware type**
Pixhawk 2.1

**Logs**

[2019-06-26 12-32-03.log](https://github.com/ArduPilot/ardupilot/files/3336454/2019-06-26.12-32-03.log)

Pedals2Paddles, 2019-06-27:
I think is likely a problem with the vehicle as opposed to the autopilot or ArduPilot firmware. The logs show ArduCopter working as intended and it doesn't stop the motors.  If the motors all stopped, I would start looking for loose connectors or wires that could affect all the ESCs (power or signal).

peterbarker, 2019-06-27:
On Thu, 27 Jun 2019, Am5x86 wrote:

> Pixhawk 2.1

Were you using a hardware "safety" switch?

Am5x86, 2019-06-27:
Hi Peter,

No, only the standard safety button on the here GPS unit.

Regards,

Simon.

From my mobile device ...

Were you using a hardware "safety" switch?

peterbarker, 2019-06-28:
On Thu, 27 Jun 2019, Am5x86 wrote:

> No, only the standard safety button on the here GPS unit.

Ah.

That's exactly the "safety" button I was refering to.

Several cases were reported where the safety switch on these units 
activated in flight, disabling motor outputs.

To work around this problem BRD_SAFETYOPTIONS was added which allows you 
to disable the safety switch in flight.  This parameter is not present in 
the firmware you were running.

http://ardupilot.org/copter/docs/parameters.html#brd-safetyoption-options-for-safety-button-behavior

> Simon.

rmackay9, 2019-06-28:
I've heard that this hardware issue with the safety switch happens most often if it gets wet.  Not sure if that was the case during this incident.  Because we've improved handling of this hardware issue with a software work-around in our latest stable I think I will close this issue.

Sorry for the loss of the vehicle, that's no fun at all.

closed 2019-06-28 by rmackay9

Am5x86, 2019-06-28:
Thanks all, did wonder on that but thought it was a software button so would see the output change on RCOUT.

Regards,

S.

From my mobile device ...

Closed #11674.

Pedals2Paddles, 2019-06-28:
The safety switch event would have shown in the log. And it would have cut the throttle output. ArduCopter was still trying to fly.

growingwings, 2019-06-28:
Which radio controller did you used?

Am5x86, 2019-07-01:
A Taranis X9D with a Frisky X8R receiver.

From my mobile device ...