ArduPilot/ardupilot #11680: Add an alt hold mode using the rangefinder

jaylthegreat, 2019-06-28:
### Feature request

**Is your feature request related to a problem? Please describe.**
Right now, there is just depth hold mode which usually just holds its depth from the surface; however, when a rangefinder is setup and healthy, it will try to maintain that rangefinder distance to the bottom rather than the depth.

**Describe the solution you'd like**
Maintaining distance from the bottom and maintaining distance from the bottom are totally distinct goals, and, as such, they should be in separate flight modes. I know that support for the rangefinder has been removed while it's being reworked, so this is a suggestion for once that is added back in.

**Describe alternatives you've considered**
The other solution might be to keep them both in depth hold mode, but then have boolean somewhere that tells the sub which distance to maintain.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ X ] Submarine

**Additional context**
[This page ](https://www.ardusub.com/operators-manual/altimeters.html) lays out that this intended for depth hold mode. 

For context, I've been working on a project and already implemented into a separate mode on my fork of the stable branch. The more tedious deal (after reworking the rangefinder) is just integrating the new mode with QGC and MavProxy

rmackay9, 2019-06-29:
As a reference, the way this is handled in copter is the range finder can be disabled using an auxiliary switch

clydemcqueen, 2024-02-22:
SURFTRAK is merged, so think we can close this bug. @Williangalvani

Williangalvani, 2024-02-23:
Done in #23435 

Thanks @clydemcqueen !

closed 2024-02-23 by Williangalvani