ArduPilot/ardupilot #11685: Buzzer is beeping continuously even after boot-up on OmnibusF7V2

ninja-zx11, 2019-06-29:
### Bug report

**Buzzer on  OmnibusF7V2 running latest daily build arducopter is buzzing continuously even after completed boot.Its connected to Buz+ and buz- pads of the board.Its not changing tones.Just continously buzzing**

**Version**
Latest daily copter build.But tried latest stable.same issue.
**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
OmnibusF7V2
[quad.txt](https://github.com/ArduPilot/ardupilot/files/3342151/quad.txt)

**Logs**
_Please provide a link to any relevant logs that show the issue_

vierfuffzig, 2019-07-01:
afaik buzzer needs a timer defined, at least that's how it works on other boards' hwdefs:
`PD15 TIM4_CH4 TIM4 GPIO(77) ALARM`

@ninja-zx11 [here's](https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/arducopter_master_OF7V2_buzzerfix.zip) copter for OF7V2 built from master with the buzzer define replaced accordingly. could you please test and report back?

ninja-zx11, 2019-07-01:
Thanks a lot vierfuffzig!! Yes it works and started to make musical tones after bootup.
I think its even better than my OmnibusF4ProV3 as buzzer on F4 beepes continously until the bootup musical tones and then it gets quiet.

But on your fixed built,it doesn't make continuous beep (like on F4Pro)while booting.It just makes musical tone after finishing the bootup.

closed 2019-07-01 by ninja-zx11

vierfuffzig, 2019-07-02:
https://github.com/ArduPilot/ardupilot/pull/11697

vierfuffzig, 2019-07-13:
fixed, so closing.