ArduPilot/ardupilot #11688: End of mission => RTL => Guided => Auto does not go back to RTL, rather....

pompecukor, 2019-06-30:
### Bug report

**Quadplane**
Undesired/unexpected functionality, probably bug.

**Issue details**

If a UAV executes last mission item (for example is a waypoint) and no proceeding command, then goes to RTL. **This is desired and expected functionality.**

**However** if during this RTL phase (while UAV is going home to land), if you switch to guided mode (this is not uncommon on surveillance mission, for example you decided to check out some POI on the way back). After returning to AUTO (pressing AUTO button on mission planner) the UAV proceeds to restart the mission from the beginning.

To make things worse, it even goes as far as ignoring sensible check like: "is flying" and "current alt > Vtol_takeoff alt".

With quadplane in the above scenario, it goes ahead and transitions to quadmode, climbs to new altitude (current + whatever is set in VTOL Takeoff) then transitions to FW and proceeds to WP2 and so on.

This can be easily tested on Mission planner SITL

I believe the expected and desired behaviors should be to go back to RTL **if** mission command marker has hit the last point. Unless "restart mission" is executed.

Again the key element here is **switching back from Guided mode,** on a VTOL that was was previously running end of mission RTL.

**Version**
Lastest master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

pompecukor, 2019-06-30:
I believe this issue may somehow be related to this older one: https://github.com/ArduPilot/ardupilot/issues/6974

IamPete1, 2021-08-29:
This can be avoided by setting the last waypoint to `RETURN_TO_LAUNCH` rather than leaving plane to switch to RTL mode once the mission is complete. 

I do think its its odd that this still reports as RTL mode, but that is a separate issue.

closed 2021-08-29 by IamPete1