ArduPilot/ardupilot #11689: SITL AutoTest for Beacons not working

peterbarker, 2019-06-30:
### Bug report

**Issue details**

A test has been added for ArduPilot's Beacon navigation: https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arducopter.py#L3351

It can be run with `./Tools/autotest/autotest.py build.ArduCopter fly.ArduCopter.BeaconPosition`

It currently fails waiting for the vehicle to be ready to arm.

```
AUTOTEST: Waiting for EKF value 271
APM: PreArm: EKF attitude is bad
APM: PreArm: AHRS not healthy
APM: EKF3 IMU0 is using range beacons
APM: EKF3 IMU0 initial pos NE = -0.2,0.1 (m)
APM: EKF3 IMU0 initial beacon pos D offset = 0.5 (m)
APM: EKF3 IMU1 is using range beacons
APM: EKF3 IMU1 initial pos NE = -0.2,0.1 (m)
APM: EKF3 IMU1 initial beacon pos D offset = 0.5 (m)
APM: PreArm: EKF attitude is bad
APM: PreArm: AHRS not healthy
APM: PreArm: EKF attitude is bad
APM: PreArm: AHRS not healthy
APM: PreArm: EKF attitude is bad
APM: PreArm: AHRS not healthy
APM: PreArm: EKF attitude is bad
APM: PreArm: AHRS not healthy
APM: PreArm: EKF attitude is bad
APM: PreArm: AHRS not healthy
```

The attitude never comes good.

This test should pass.  Once it's fixed it can be removed from the list of disabled tests and it will ensure we won't break navigate-by-beacon (at least in SITL...)

**Version**

master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
+

**Hardware type**
SITL

**Logs**

n/a

rmackay9, 2020-05-12:
So one thing that is missing from the auto test is we need to set AHRS_EKF_TYPE = 3.  This doesn't resolve all the issue though I'm afraid.  At least I can confirm that the autotest for Beacons is not working.  It may be that we've broken beacon support in EKF3.

peterbarker, 2020-11-25:
Test was fixed.

closed 2020-11-25 by peterbarker