ArduPilot/ardupilot #11743: RevoMini wrong IMU rotation on hwdef.dat

anbello, 2019-07-08:
### Bug report

**Issue details**

After [this](https://github.com/ArduPilot/ardupilot/commit/614a059819855f217b0a5ad111c549fccac06236#diff-11d8dd1fa26a0cff3376c94acb0fd570) commit the IMU has a wrong orientation in hwdef.dat, this line:
`IMU Invensense SPI:mpu6000 ROTATION_YAW_90`
should be:
`IMU Invensense SPI:mpu6000 ROTATION_YAW_180`

**Version**
master

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
I tested on X quadcopter

**Hardware type**
RevoMini

**Logs**

anbello, 2019-07-08:
I did a PR for this:
https://github.com/ArduPilot/ardupilot/pull/11744

vierfuffzig, 2019-07-15:
thanks andrea, fix is merged, so closing this.

closed 2019-07-15 by vierfuffzig