ArduPilot/ardupilot #11756: ArduPilot for Fixed-Wing about Guided mode

zhanghouxin07, 2019-07-09:
Hi everyone,
I'm a newbie about ardupilot. I want to use an external computer（tx2 or intel nuc） to control fixed-wing flight (real-time sending of waypoints, or control gestures) through the mavros protocol. I have taken a cursory look at the ardupilot user manual. I discover the guide mode.

so, the GUIDE mode can be utilize for controlling fixed-wing flight in real-time?
For instance,  after the fixed-wing takes off by RC, the ardupilot enter the GUIDE mode and receive the waypoints(or other cmds) from external computer(tx2). 
and then the fw filght mode can be switch to RC again.

thanks a lot, everyone.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

zhanghouxin07, 2019-07-09:
Does ardupoilt also have offboard mode?

auturgy, 2019-07-09:
No. Guided mode (ArduPilot) and offboard mode (px4) are equivalent in many ways - such as interfacing with a companion computer via an api (DroneKit, mavros, etc). Offboard mode is not available for fixed wing though, whilst Guided mode is.
Support questions are best asked on discuss.ardupilot.org

Regards,

James

zhanghouxin07, 2019-07-09:
> No. Guided mode (ArduPilot) and offboard mode (px4) are equivalent in many ways - such as interfacing with a companion computer via an api (DroneKit, mavros, etc). Offboard mode is not available for [...]

I want to confirm, that is, ardupilot can realize real-time control of fixed-wing flight with an external computer in Guide Mode?

peterbarker, 2019-07-09:
On Tue, 9 Jul 2019, houxin zhang wrote:

> I want to confirm, that is, ardupilot can realize real-time control of
> fixed-wing flight with an external computer in Guide Mode?

See this autotest:

https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arduplane.py#L273

peterbarker, 2020-11-25:
I think this was adequately answered.

closed 2020-11-25 by peterbarker