ArduPilot/ardupilot #11758: Add way to limit range to meet UAV export requirements

tridge, 2019-07-09:
For companies wanting to export vehicles they need a way to limit the maximum range to a maximum of 300km. This is a requirement of the Wassenaar agreement and applies to many (most?) countries.

I propose to implement as follows:
 - extend AFS to add a AFS_RANGE_MAX parameter in kilometers
 - ensure actions for fixed wing, quadplane and copters allow for forced landing on AFS breach
 - for origin of the radius use the first 3D lock GPS position at the point that the vehicle is first armed
 - add ability to lock parameters set using apjtool, so they can't be modified using a GCS. Likely using a READONLY flag on the end of the parameter line

duran9dv, 2019-07-09:
possible warning Xkm before the 300 limit to the user, ie at 200km warning, 250km effected and lands

OXINARF, 2019-07-09:
> add ability to lock parameters set using apjtool, so they can't be modified using a GCS. Likely using a READONLY flag on the end of the parameter line

This would be useful, independently of this feature. One thing to be careful though is that we want to still be able to modify the parameters from code, just not from user input.

tridge, 2019-07-09:
> This would be useful, independently of this feature. One thing to be careful though is that we want to still be able to modify the parameters from code, just not from user input.

yes, that was the plan

peterbarker, 2019-07-09:
On Tue, 9 Jul 2019, Francisco Ferreira wrote:

> add ability to lock parameters set using apjtool, so they can't
>       be modified using a GCS. Likely using a READONLY flag on the end
>       of the parameter line
>
> This would be useful, [...]

That might make scripting a little interesting.

closed 2021-08-17 by tridge