ArduPilot/ardupilot #11783: Provide Sag compensated voltage rather than raw battery voltage to mavlink

lthall, 2019-07-12:
It would be nice to be able to see the sag compensated voltage from the ground station. I suspect we don't have the data structure to do this separately so we may need to provide a parameter based choice.

One option might be to provide the sag compensated voltage if we are using that voltage to trigger the failsafe but I think a separate parameter would be more effective.

aliakbarkhan, 2022-02-02:
> It would be nice to be able to see the sag compensated voltage from the ground station. I suspect we don't have the data structure to do this separately so we may need to provide a parameter based [...]

hi ithall i would like to contribute to this issue as it is my first time in open source could you please elaborate a little and guide me in the right path

hendjoshsr71, 2022-02-03:
Hi @aliakbarkhan 

Let me explain a bit what you will likely need to do.

There is a parameter that tells the battery_failsafe system to use the sag compensated voltage for the failsafe rather than the raw (albeit filtered) voltage, [here](https://github.com/ArduPilot/ardupilot/blob/d73a45aa645fd6d08f41d234679dd1bf0b1ec21f/libraries/AP_BattMonitor/AP_BattMonitor_Params.cpp#L62-L67).  The sag voltage estimates the voltage drop due to the battery's internal resistance and  then adds that estimate back to the raw voltage. 

Reporting of the battery voltage to the GCS happens through  either the BATTERY_STATUS mavlink message here, 
`void GCS_MAVLINK::send_battery_status(const uint8_t instance) const` or `void GCS_MAVLINK::send_sys_status()`  I forget which is used by each gcs.

You will somewhere want to use this method instead to fill the voltage field dependent upon the parameter setting   
`float AP_BattMonitor::voltage_resting_estimate(uint8_t instance) const`

so in `float AP_BattMonitor::voltage(uint8_t instance) const` you would instead use the resting estimate. 

That will get you a good part of the way there. (Assuming we dont have to deal with the `has_cell_voltages()` case)

To test this you will need to use SITL, setup a battery monitor, and try the parameter on the different ground control stations)

aliakbarkhan, 2022-02-04:
tysm josh for even writing this much, i understood a little. So the paramter sag compensated voltage is available i have to just display it to the GSC  through BATTERY_STATUS mavlink .I didn't understood (**You will somewhere want to use this method instead to fill the voltage field dependent upon the parameter setting
float AP_BattMonitor::voltage_resting_estimate(uint8_t instance) const

so in float AP_BattMonitor::voltage(uint8_t instance) const you would instead use the resting estimate.**) what did u mean in this and one last think can too tell how to setup a battery monitor im using cygwin,mavlink&mission planner that would be really helpfull

aliakbarkhan, 2022-02-12:
> void GCS_MAVLINK::send_sys_status()

can you help me a little more please

hendjoshsr71, 2022-03-22:
Closed via #20332

closed 2022-03-22 by hendjoshsr71