ArduPilot/ardupilot #11785: Copter: AP_Motors's roll factors for OctoV seem incorrect

rmackay9, 2019-07-12:
The roll factors for the OctoV ([code](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMatrix.cpp#L632), [wiki](http://ardupilot.org/copter/docs/connect-escs-and-motors.html#octocopter)) seem incorrect.

Below is an image showing what the roll and pitch factors are for each motor at the moment in master.
[image]

Master's Motor, Roll Factor, Pitch Factors
**1**,1.0f,0.34f
**2**, -1.0f,-0.32f
**3**,1.0f,-0.32f
**4**,-0.5f,-1.0f
**5**,1.0f,1.0f
**6**,-1.0f,0.34f
**7**,-1.0f,1.0f
**8**,0.5f,-1.0f

The pitch factors look like they could be correct (or close to correct) although I would have expected that motors 2 and 6 would have the same factor except reversed (and same thing for motors 1 and 3).

The roll factors look very incorrect.  I would have expected the front motors factor to be 1 (which it is) and then each subsequent pair of motors (going backwards) to have lower roll-factors because they are closer to the frame.

Proposed Motor, Roll Factor, Pitch Factors
**1**,0.83f,0.34f
**2**, -0.67f,-0.32f
**3**,0.67f,-0.32f
**4**,-0.5f,-1.0f
**5**,1.0f,1.0f
**6**,-0.83f,0.34f
**7**,-1.0f,1.0f
**8**,0.5f,-1.0f

This is discussed on rcgroups here: https://www.rcgroups.com/forums/showthread.php?2768760-Building-Flying-a-V-octocopter

As a side note, the motor order diagrams for the OctoV are also incorrect: https://github.com/ArduPilot/ardupilot_wiki/issues/1838

lthall, 2019-07-12:
[image]

lthall, 2019-07-12:
Those coefficients are completly crazy. This is what it should be for the rear props being the same spacing as the side props.

[image]

rmackay9, 2019-07-12:
@lthall, great, thanks very much for the review!  I plan to merge [the fix](https://github.com/ArduPilot/ardupilot/pull/11788) right after the person who brough this to my attention tests it on a real vehicle.

rmackay9, 2019-07-18:
fixed in mater so closing, txs!

closed 2019-07-18 by rmackay9